rwaldron/johnny-five #1264: Docs: Relays page links to invalid page

evancohen, 2016-12-24:
**Where**
http://johnny-five.io/api/relays/
> For working with a single relay check out the [Relay](http://johnny-five.io/api/relays/relay) class.

**What**
A broken relative link points to an invalid URL for the Relay class.

Happy to send a PR to fix this. Couldn't find this content in the `docs/` folder, is there a docs repo that I can send it to?

scottgonzalez, 2016-12-24:
You can fix it in the [wiki](https://github.com/rwaldron/johnny-five/wiki/Relays) and that will make it to the web site.

evancohen, 2016-12-24:
If only it were that easy! It seems that the link works in the wiki but not on the documentation site. 

I could change the link to `../relay` (or `/api/relay`) so it would work on the site, but then it would break in wiki. Not sure what the correct course of action is here.

scottgonzalez, 2016-12-24:
Why would `../relay` not work on the wiki?

FWIW, this looks like it's probably broken for all collections, e.g., Servos have the same problem.

scottgonzalez, 2016-12-24:
Looks like `Boards` is almost correct. It uses a full wiki URL (https://github.com/rwaldron/johnny-five/wiki/Board) to link to `Board`. On the site, this results in `/api/Board`, which would work if `Board` were lowercase.

dtex, 2018-12-31:
Just checked. These are all working on the wiki so it was fixed at some point.

closed 2018-12-31 by dtex