rwaldron/johnny-five #1265: to get raw data of Compass

hungrykirby, 2016-12-28:
excuse me.
i want to get raw data of compass.
so i would like you to change some code.

i temporary add this to my packege at compass.js.
`
raws:{
          get:function(){
              return raw;
          }
      },
`
if i can get raw data in another way, could you please tell me. Thanks!

dtex, 2018-01-02:
It's been a while, and I'm surprised nobody else has responded. I apologize for that.

I took a quick look at some other sensor classes and they all (except for the generic Sensor) don't have the option to return raw values. Can I ask your use case for knowing the raw value instead of the heading or bearing?

dtex, 2018-01-16:
I'm gonna close this since there has been no activity on the ticket. Feel free to re-open. It doesn't seem unreasonable to me that raw values should be available for all devices, I just want to make a strong case.

closed 2018-01-16 by dtex