rwaldron/johnny-five #1266: Johnny-Five & Bleno collision - Raspberry Pi 3

yakeer, 2016-12-30:
Hey,
I'm trying to implement both Johnny Five, and Bleno framework together.
Once I init the board, the hci0 seems unresponsive, and I cannot get it to work again until I reboot the pi.

Anyone has ever encountered this collision/issue?
Any fix?

don, 2017-02-17:
@yakeer it looks like raspi-serial is causing a problem

https://github.com/nebrius/raspi-io/blob/f97acd085e70cfa93075a1d80cf5b5be5fb201f9/dist/index.js#L19

don, 2017-02-17:
@yakeer change the raspi-serial DEFAULT_PORT to /dev/ttyS0 since /dev/ttyAMA0 is mapped to Bluetooth on the RPi 3. (@sandeepmisty tracked that down.)

https://github.com/nebrius/raspi-serial/blob/823eb9470c393a6e7487fe9c3bedc12a35302dc1/dist/index.js#L37

rwaldron, 2017-03-30:
@nebrius ping

nebrius, 2017-03-30:
Yep: https://github.com/nebrius/raspi-serial/issues/3

nebrius, 2017-04-07:
I just published v7.3.0 of raspi-io, which now disables serial support in Raspi IO on the RPi 3 by default, and I included steps on how to enable it in the wiki.

Note: `/dev/ttyS0` isn't a full-fledged UART, it's this funky software emulated thing. As such, I felt it prudent to just disable serial and leave it up to users to decide if they want proper serial or bluetooth, because unfortunately it's not possible to have both on the RPi 3 :(

sandeepmistry, 2017-04-10:
@nebrius awesome!

I think the new Raspberry Pi Zero W will have the same problem as the RPi 3, did you enable the same behaviour for it as well?

nebrius, 2017-04-10:
> I think the new Raspberry Pi Zero W will have the same problem as the RPi 3, did you enable the same behaviour for it as well?

No I didn't. I don't have a Raspberry Pi Zero W in my possession yet, so I can't get the board revision. Do you (or anyone else reading) have one on hand by chance? If so, can you run `cat /proc/cpuinfo` on it and paste the output here?

sandeepmistry, 2017-04-10:
Yes, here you go:

```
pi@raspberrypi-zero-w:~ $ cat /proc/cpuinfo 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 997.08
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2708
Revision	: 9000c1
Serial		: 0000000018b693a9
```

`hciattach` is configured for `/dev/serial1` which maps to `/dev/ttyAMA0` like the Raspberry Pi 3.

```
pi@raspberrypi-zero-w:~ $ ps -eaf | grep hciattach
root       503     1  0 23:10 ?        00:00:00 /usr/bin/hciattach /dev/serial1 bcm43xx 3000000 flow -
pi         670   651  0 23:13 pts/0    00:00:00 grep --color=auto hciattach
pi@raspberrypi-zero-w:~ $ ls -l /dev/serial1 
lrwxrwxrwx 1 root root 7 Mar 25 02:14 /dev/serial1 -> ttyAMA0
```

Let me know if you need anymore details.

mwittig, 2017-04-13:
> I just published v7.3.0 of raspi-io, which now disables serial support in Raspi IO on the RPi 3 by default,  and I included steps on how to enable it in the wiki.

@nebrius Thanks so much. Seems like this is the solution to the issue I have filed earlier. https://github.com/nebrius/raspi-io/issues/83

nebrius, 2017-04-28:
Following up, this should be addressed in Raspi IO 8.0.0. Let me know if it works for you!

sandeepmistry, 2017-05-15:
The upstream changes look good to me! @nebrius thanks for looking into this!

dtex, 2018-01-02:
It looks like @nebrius addressed these issue in rasp-io v7.3.0 and v8.0.0 so I'm going to close this. Pleaes re-open if I have missed something.

closed 2018-01-02 by dtex