rwaldron/johnny-five #1267: Which is the supported node version?

fabianmoronzirfas, 2017-01-06:
I there, I'm trying to revive an old project and was wondering which Node version is the currently supported one? While doing a quick test with nvm, everything else then v0.12.x failed on macOS 10.12.2 due to some error with the serialport package and something with node-pre-gyp.  

It would be great if the docs or the README could state the latest confirmed versions.

fabianmoronzirfas, 2017-01-06:
Hm. Okay. This is what happened in Node v7.4

This is my package.json
```json
{
  "name": "blink",
  "version": "1.0.0",
  "description": "blink it",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1",
    "start": "node index.js"
  },
  "repository": {
    "type": "git",
    "url": "git@github.com:fabiantheblind/johnny-five-is-alive.git"
  },
  "author": "",
  "license": "ISC",
  "dependencies": {
    "express": "^4.13.0",
    "johnny-five": "^0.8.81"
  }
}
```
Running the install command in my project root

    npm install

gives me this error:  

```shell
> serialport@1.7.4 install /Users/icke/Documents/johnny-five-is-alive/blink/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! UNCAUGHT EXCEPTION
node-pre-gyp ERR! stack Error: Cannot find module 'internal/fs'
node-pre-gyp ERR! stack     at Function.Module._resolveFilename (module.js:470:15)
node-pre-gyp ERR! stack     at Function.Module._load (module.js:418:25)
node-pre-gyp ERR! stack     at Module.require (module.js:498:17)
node-pre-gyp ERR! stack     at require (internal/module.js:20:19)
node-pre-gyp ERR! stack     at evalmachine.<anonymous>:18:20
node-pre-gyp ERR! stack     at Object.<anonymous> (/Users/icke/Documents/johnny-five-is-alive/blink/node_modules/serialport/node_modules/node-pre-gyp/node_modules/tar/node_modules/fstream/node_modules/graceful-fs/fs.js:11:1)
node-pre-gyp ERR! stack     at Module._compile (module.js:571:32)
node-pre-gyp ERR! stack     at Object.Module._extensions..js (module.js:580:10)
node-pre-gyp ERR! stack     at Module.load (module.js:488:32)
node-pre-gyp ERR! stack     at tryModuleLoad (module.js:447:12)
node-pre-gyp ERR! System Darwin 16.3.0
node-pre-gyp ERR! command "/usr/local/Cellar/node/7.4.0/bin/node" "/Users/icke/Documents/johnny-five-is-alive/blink/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /Users/icke/Documents/johnny-five-is-alive/blink/node_modules/serialport
node-pre-gyp ERR! node -v v7.4.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! This is a bug in `node-pre-gyp`.
node-pre-gyp ERR! Try to update node-pre-gyp and file an issue if it does not help:
node-pre-gyp ERR!     <https://github.com/mapbox/node-pre-gyp/issues>
```

I still tried to execute the index.js and got this:  

```shell
module.js:472
    throw err;
    ^

Error: Cannot find module 'serialport'
    at Function.Module._resolveFilename (module.js:470:15)
    at Function.Module._load (module.js:418:25)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Board.detect (/Users/icke/Documents/johnny-five-is-alive/blink/node_modules/johnny-five/lib/board.js:70:7)
    at new Board (/Users/icke/Documents/johnny-five-is-alive/blink/node_modules/johnny-five/lib/board.js:328:23)
    at Object.<anonymous> (/Users/icke/Documents/johnny-five-is-alive/blink/app/j5.js:3:13)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
```

Just for fun I ran a

npm install serialport (also gives a lot of errors but seems to install something)

then the error for `node index.js` is

```shell
module.js:472
    throw err;
    ^

Error: Cannot find module 'firmata'
    at Function.Module._resolveFilename (module.js:470:15)
    at Function.Module._load (module.js:418:25)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Board.connect (/Users/icke/Documents/johnny-five-is-alive/blink/node_modules/johnny-five/lib/board.js:142:7)
    at Board.<anonymous> (/Users/icke/Documents/johnny-five-is-alive/blink/node_modules/johnny-five/lib/board.js:329:26)
    at Board.<anonymous> (/Users/icke/Documents/johnny-five-is-alive/blink/node_modules/johnny-five/lib/board.js:135:16)

```

after that I ran a 

    npm install firmata

Then everything worked fine. 🤔

closed 2017-01-06 by fabianmoronzirfas

rwaldron, 2017-01-06:
Weird 0_o