rwaldron/johnny-five #1268: Johnny-Five hangs forever on exit on Raspberry Pi 3

nemesszili, 2017-01-08:
I have been trying for a day to configure an optimal environment for running Johnny-Five code on my Raspberry Pi 3. While the code actually gets executed, for some reason I can't exit the app in a clean manner. More specifically, when I type ".exit" or press ^C twice, it prints _Board Closing_, but it hangs forever and I need to do a kill -9 on its PID to make it quit. Here are the steps that I took:

1. Clean Raspbian flash (2016-11-25 image)
2. Installed NodeJS
```
sudo apt-get update
sudo apt-get install -y curl apt-transport-https ca-certificates &&
curl --fail -ssL -o setup-nodejs https://deb.nodesource.com/setup_6.x &&
sudo bash setup-nodejs &&
sudo apt-get install -y nodejs build-essential

node -v: 6.9.4
npm -v: 3.10.10
```
3. Created a directory and installed johnny-five and raspi-io
`npm install johnny-five raspi-io`
_Please note that I did this WITHOUT sudo!_
4. Entered some code to test my relay
```
var five = require("johnny-five");
var Raspi = require("raspi-io");
var board = new five.Board({
  io: new Raspi()
});

board.on("ready", function() {
  var relay = new five.Relay("P1-7");
  relay.off();
  relay.on();
});
```
5. Did a reboot
6. Ran the code (WITH sudo)
`sudo node johnny.js`
7. Tried to exit
```
>> .exit
1483871185654 Board Closing.  
```
8. ...still hanging

Could anyone retrace my steps and confirm the existence of this bug or suggest a new way to approach the issue?

rwaldron, 2017-01-09:
@nebrius I'm not sure how to help with this, do you have any thoughts?

nebrius, 2017-01-09:
That's...weird...maybe something is getting registered for `beforeExit` and is preventing Node from exiting cleanly? I'll try and do some digging when I get some free time. I made a change to one of the underlying raspi-io libraries that _may_ be related.

nebrius, 2017-01-10:
There indeed was a weird bug. I'm not sure what was causing it, but something with the REPL in Johnny-Five and the interrupt routine registered in raspi-io was conflicting somehow causing the process to not respond to kill signals. I wrote up a workaround in raspi-io so that it doesn't hang anymore. Check out v7.2.2 of raspi-io.

nemesszili, 2017-01-10:
Guys, thanks a lot for the prompt response. It works like a charm!

closed 2017-01-10 by nemesszili

markjohnson303, 2018-01-25:
@nebrius I'm experiencing this same issue.

nebrius, 2018-01-31:
@markjohnson303 I just tested with the latest version of everything and it worked fine. Can you try upgrading to Johnny-Five 0.13.0 and Raspi IO 8.1.0 and see if you still get that problem?