rwaldron/johnny-five #1271: use esp8266-12E,  connect to DS18B20, bur read error.

chedongzhe, 2017-01-12:
I want to use esp8266-12E, to connetion DS18B20(temperature sensor)，use serialport communication, but read error in johnny-five.

1. esp8266-12E (Using ConfigurableFirmata, port speed baudrate: 9600).
2. johnny-five program (Node version 7.4. System: Windows 10) :
   var five = require("johnny-five");
   var SerialPort = require("serialport").SerialPort;
   var board = new five.Board({
       port: new SerialPort("COM6",{
         baudrate: 9600
   });
   board.on("ready", function () {
      var temp = new five.Thermometer({
          controller: "DS18B20",
          pin: 13
   });
   temp.on("change", function () {
      console.log(this.celsius + "°C");
   });
   });

3. but error message:

D:\Project\johnnyFiveProject>node johnnyfive.js
1484179244173 SerialPort COM6
1484179244178 Connected COM6
1484179254181 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
events.js:165
      throw err;
      ^

Error: Uncaught, unspecified "error" event. ([object Object])
    at Board.emit (events.js:163:17)
    at Board.log (C:\Users\CheDongZhe\AppData\Roaming\npm\node_modules\johnny-five\lib\board.js:630:8)
    at Board.(anonymous function) [as error] (C:\Users\CheDongZhe\AppData\Roaming\npm\node_modules\johnny-five\lib\board.js:641:14)
    at Board.<anonymous> (C:\Users\CheDongZhe\AppData\Roaming\npm\node_modules\johnny-five\lib\board.js:390:14)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)

soundanalogous, 2017-01-12:
The problem is that pin 1 on the ESP8266 interferes with serial in Firmata. The solution is to modify ConfigurableFirmata to ignore pin 1.

User firmatabuilder.com to generate a version of Firmata using Serial, 9600 baud (you may actually be able to use 57600 actually since I believe ESP auto-detects baud rate - or maybe that is something that Adafruit enabled specifically for their Huzzah ESP8266 board). Then in the `setup()` function, add the line `Firmata.setPinMode(1, PIN_MODE_IGNORE);` as shown below:

```c++
void setup()
{
  initFirmata();

  initTransport();

  Firmata.setPinMode(1, PIN_MODE_IGNORE);

  systemResetCallback();
}
```

chedongzhe, 2017-01-12:
1. ConfigurableFirmata code.
   . . . 
   #include <OneWireFirmata.h>
   OneWireFirmata oneWire;
   #include <SerialFirmata.h>
   SerialFirmata serial;
   . . . 
   void initTransport()
   {
        Firmata.begin(9600);
   }
  . . . . . 
   void setup()
   {
        initFirmata();
        initTransport();
        Firmata.setPinMode(1, PIN_MODE_IGNORE);    //added, compile and upload to esp8266-12E is OK
        systemResetCallback();
  }

2. johnny-five code.
      var five = require("johnny-five");
      var SerialPort = require("serialport");
      var board = new five.Board({
            port: new SerialPort("COM6",{
                baudrate: 9600
            })
      });
     board.on("ready", function () {
         var temp = new five.Thermometer({
             controller: "DS18B20",
             pin: 13
         });
        temp.on("change", function () {
            console.log(this.celsius + "°C");
        });
    });
3. run johnny-five code,  erro message:
1484189036400 SerialPort COM6
1484189036404 Connected COM6
1484189046407 Device or Firmware Error A timeout occurred while connecting to the Board.
Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
events.js:165
      throw err;
      ^

Error: Uncaught, unspecified "error" event. ([object Object])
    at Board.emit (events.js:163:17)
    at Board.log (C:\Users\CheDongZhe\AppData\Roaming\npm\node_modules\johnny-five\lib\board.js:630:8)
    at Board.(anonymous function) [as error] (C:\Users\CheDongZhe\AppData\Roaming\npm\node_modules\johnny-five\lib\board.js:641:14)
    at Board.<anonymous> (C:\Users\CheDongZhe\AppData\Roaming\npm\node_modules\johnny-five\lib\board.js:390:14)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)

soundanalogous, 2017-01-12:
Try not including SerialFirmata in your ConfigurableFirmata build.

soundanalogous, 2017-01-12:
Also try 57600, 9600 may be too slow for Johnny-five, or you may have to extend the Johnny-five timeout (I forget how to do that off the top of my head... it's an object you pass to the `five.Board` constructor).

As a side, the ESP board should work fine over Wi-Fi instead of Serial so you could try that if Wi-Fi works for your environment (sometimes if you are in university it may not if the Wi-Fi policy has certain regulations).

soundanalogous, 2017-01-12:
Also, I have no idea if the ESP boards are compatible with the OneWire library. You may want to test that in a regular Arduino sketch first.

chedongzhe, 2017-01-12:
1. try baudrate  57600 or  try  EtherPort, the same error message.

2. if I use  OneWire library(with Platformio). to test ds18b20 sensor (pin 13) is ok.

soundanalogous, 2017-01-12:
Have you tried to get something simple working like blinking an LED rather than jumping straight into a OneWire device?

soundanalogous, 2017-01-12:
Also these are the only ESP8266 boards that have been proven to work with Firmata: https://github.com/firmata/arduino/wiki/WiFi-compatible-boards-and-shields#esp8266. It's possible some other boards may not work for unknown reasons.

chedongzhe, 2017-01-12:
only OneWire

soundanalogous, 2017-01-12:
I forgot earlier, but you also need to ignore pin 3. Here's a StandardFirmata example for using an ESP8266 over Serial. This does work, at least for an Adafruit Huzzah ESP8266 board: https://gist.github.com/soundanalogous/87ea10c07c3f072cb46d8cad206b42d1. I didn't have access to this earlier since I was using a different computer.

chedongzhe, 2017-01-12:
added  : 
   Firmata.setPinMode(1, PIN_MODE_IGNORE);
   Firmata.setPinMode(3, PIN_MODE_IGNORE);

but the same error.

soundanalogous, 2017-01-12:
I suspect your ESP board is unfortunately not compatible then.

soundanalogous, 2017-01-12:
Okay I spend a little more time on this, testing locally with my Huzzah 8266 board. I'm not able to get the db18b20 to work, but I can get an I2C digital compass to work with the same setup (both with Serial and with Wi-Fi), so the issue is definitely somewhere in the OneWire implementation in one of the many layers of the stack. It could be in any of the following (from the lowest to the highest level):

1. The [OneWire library](https://github.com/PaulStoffregen/OneWire/issues/14)
2. The [OneWireFirmata class](https://github.com/firmata/ConfigurableFirmata/blob/master/src/OneWireFirmata.cpp) in ConfigurableFirmata
3. The OneWire Firmata implementation in firmata.js
4. The db18b20 implementation in [johnny-five](https://github.com/rwaldron/johnny-five/issues/1115).

Since you say it works fine outside of Firmata, we can probably rule out #1, but when you compile ConfigurableFirmata the bundled version of OneWire is used which is slightly older than the version you get with Platform.IO.

Regarding #2. The OneWire wrapper in ConfigurableFirmata is not well supported. The original author is no longer involved with the project and I don't have a strong understanding of OneWire to properly maintain it. No one else has stepped in to provide support.

Regarding #3. The firmata.js implementation is only as good as the ConfigurableFirmata implementation.

chedongzhe, 2017-01-14:
1. ESP8266-12E (WIFI).  code added.  
  . . . 
  Firmata.setPinMode(1, PIN_MODE_IGNORE);
  Firmata.setPinMode(3, PIN_MODE_IGNORE);
 . . .
2. johnny-five javascript code
   . . .
    var board = new five.Board({
       port: new EtherPort(3030)
    });
   . . .
3. run johnny-five code then error message : 

1484371639990 SerialPort Listening on port: 3030
1484371639994 Connected Listening on port: 3030
1484371645066 Repl Initialized
>> events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: read ECONNRESET
    at exports._errnoException (util.js:1022:11)
    at TCP.onread (net.js:572:26)

soundanalogous, 2017-01-14:
In order to use the ESP8266 over Wi-Fi, you need to compile and upload [StandardFirmataWiFi](https://github.com/firmata/arduino/tree/master/examples/StandardFirmataWiFi) to the board. By default StandardFirmataWiFi is configured as a TCP server so you'd have to use [etherport-client](https://www.npmjs.com/package/etherport-client) rather than [etherport](https://www.npmjs.com/package/etherport). However, you can configure as a Wi-Fi client instead by entering the IP address of the computer running the J5 application. You enter that [here](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataWiFi/wifiConfig.h#L114). Read the instructions starting from the top of wifiConfig.h for more details: https://github.com/firmata/arduino/blob/master/examples/StandardFirmataWiFi/wifiConfig.h#L114.

Also be sure to enter your network name and passphrase in wifiClient.h.

You can find johnny-five esp client and server examples here: https://github.com/soundanalogous/firmata-client-examples/tree/master/johnny-five/network.

closed 2019-01-07 by stale[bot]