rwaldron/johnny-five #1272: Can I use an ultrasonic sensor with Johnny-five?

zoooofu, 2017-01-13:
I would like to add an ultrasonic sensor to achieve obstacle avoidance, is it possible to use Johnny-five to do this? Unfortunately, I can't find any api about it.

fivdi, 2017-01-15:
Yes, it is possible. The API is described [here](http://johnny-five.io/api/proximity/). It's important to point out that a board needs to support the "PING_READ" extension for HC-SR04 ultrasonic sensors to function correctly. Whether or not "PING_READ" is supported by a specific board can be seen in the feature tables for the various boards [here](http://johnny-five.io/platform-support/). For "PING_READ" support. PingFirmata will be required on many of the boards

rwaldron, 2017-01-16:
Thanks @fivdi!

closed 2017-01-16 by rwaldron

zoooofu, 2017-02-06:
@fivdi All right, thanks!