rwaldron/johnny-five #1273: DHT11  (via i2C nano backpack)

lezabour, 2017-01-13:
hey
really having a hard time with this ;)
I have the DHT11 PCB version (https://www.aliexpress.com/w/wholesale-dht11.html)
Work good directly on the arduino uno(not johnny five).
I used a arduino nano, to make like you describe the I2C nano backpack.

The first problem was to put the firmware on the nano backpack. the dht11_i2c_nano_backpack.ino is calling a DHT.h file, 
I found it on internet (via adafruit), and finally compiled and send dht11_i2c_nano_backpack.ino to the nano backpack

Now it's done, but i still have no data 
temperature
  celsius           :  0
  fahrenheit        :  32
  kelvin            :  273.15
--------------------------------------

I used this code:

`board.on("ready", function() {
	console.log('rdy');
  var multi = new five.Multi({
    controller: "DHT11_I2C_NANO_BACKPACK"
  });

setInterval(function() {
	

    console.log("temperature");
    console.log("  celsius           : ", multi.temperature.celsius);
    console.log("  fahrenheit        : ", multi.temperature.fahrenheit);
    console.log("  kelvin            : ", multi.temperature.kelvin);
    console.log("--------------------------------------");

    console.log("Hygrometer");
    console.log("  relative humidity : ", multi.hygrometer.relativeHumidity);
    console.log("--------------------------------------");
    }, 5000);
});
`
I dont know if i need to use a resitor (used it anyway) since i have the PCB version...

thanks for help ;)

lyzadanger, 2017-01-23:
Hi, @lezabour !

Could you tell me more about your setup? What kind of board is the Nano (backpack) currently connected to physically? Is it connected to an Arduino Uno? A diagram or photo of your circuit/setup would be super helpful if you are able, also. Do you have something akin to [http://johnny-five.io/api/multi/#dht11i2cnanobackpack](this configuration)?

lezabour, 2017-01-24:
Hi @lyzadanger,
Thanks for your answer ;)
I'm using: 
Nano V3.0 - ATmega 328P  as backpack  (with the dht11_i2c_nano_backpack.ino firmware)
An arduino UNO as main controller (with Firmata on it)
All is wired like in the schematic here [http://johnny-five.io/api/multi/#dht11i2cnanobackpack]
2 difference with this schematic: 
- the DHT11 is the PCB version (so it already has a resistor on it) -> [https://www.aliexpress.com/w/wholesale-dht11.html]
- the resistor is not the same (i use 1kOhm resistor, as it's not describe which one we need)
At the end, all this is plug to a raspberry pi 2, containing nodejs / johnny five etc....

So i have no data coming (always 0). I tryed:
- Maybe it's a sensor problem, i switch with another DHT11, same result.
- Maybe because i have the DHT11 PCB, i don't need resistor, tryed without it, same result.

Really don't know what do to

Thanks for your help

martip-sainsburys, 2017-03-03:
I'm seeing the same thing, I have the same chip but with 3 connectors ( https://www.aliexpress.com/item/1pcs-free-shipping-DHT11-Temperature-And-Relative-Humidity-Sensor-Module-PCB-for-Arduino-DIY-Starter-Kit/32631854864.html )

Also seeing temp always 0 as well as humidity.

felansu, 2017-03-04:
@martip-sainsburys exactly same problem !

rwaldron, 2017-03-30:
Has everyone here tried the other, newer DHT* firmware? It's also available here: https://github.com/rwaldron/johnny-five/tree/master/firmwares, in https://github.com/rwaldron/johnny-five/blob/master/firmwares/dht_i2c_nano_backpack.ino . To be clear, I have no idea if this will fix your issue, but worth a try to see what we can learn. The firmware is just using the Arduino DHT library. Can someone share the arduino C sketch that works correctly? I could just use the one given in the Arduino IDE examples, but I want to make sure I'm comparing apples to apples :)

dtex, 2018-01-16:
@felansu @martip-sainsburys and @lezabour 

I just want to followup and see if any of y'all were able to try the other firmware that @rwaldron suggested.

drprime, 2018-05-15:
Have thks problem to with dht22. 
Trying Uno+nano, mega + nano.  Celsium temperature is 0
firware dht_i2c_nano_backpack.ino
Without johny-five working fine(test scatch from dht lib)
And i have another question. How to connect 2 or more (i need 6) sensors using johny-five?

NguyenHoangSon96, 2019-09-05:
So what is the solution guys, i have the same proplem

marti2221, 2019-11-05:
I have the same problem as well, would love to hear some more ideas

TooFast4ya, 2019-11-10:
Same for me :(

EduAir, 2020-08-28:
same for me ! Any idea?

derplarsen, 2021-09-27:
This actually works.. i noticed the values were correct when you subract 1000 from C, 1800 from F, and 1000 from K :) 🤷‍♂️
```
  temperature.on("data", function() {
    console.log("celsius: %d", this.C-1000);
    console.log("fahrenheit: %d", this.F-1800);
    console.log("kelvin: %d", this.K-1000);
  });
```