rwaldron/johnny-five #1275: Feather HUZZAH (ESP8266), generic H-bridge motor drivers, and PWM range

hxlnt, 2017-01-14:
I recently led a series of NodeBots workshops using J5 + StandardFirmataWifi with the Adafruit Feather HUZZAH boards and inexpensive dual H-bridge motor drivers (L9110, various manufacturers). I'd like to use the learnings from preparing and hosting this workshop to help improve J5's Feather support. 

In particular, I initially ran into issues with adjusting the speed and direction of DC motors due to the [ESP8266's default PWM range of 0 to 1023](http://arduino.esp8266.com/versions/1.6.5-1160-gef26c5f/doc/reference.html#analog-output) instead of 0 to 255. Passing a speed variable to the motors resulted in erratic results, and motor.rev() functions did not work regardless of what speed was passed in. In our workshop, the issue was corrected by manually changing the PWM range on the Firmata side (`analogWriteRange(255)`), but I would be interested to see if this would be worth addressing in J5.

I am happy to help write the code, but this seems like something that might have broader implications than just the Feather board + L9110 motor drivers (_e.g.,_ possible implementation of PWM range parameter in all ESP8266 boards?), and I don't have a good handle on J5's overall philosophy on changes like this one.

If it's useful to anyone in the meantime, the source code and step-by-step instructions for the Feather NodeBots workshop are [on GitHub](https://github.com/hxlnt/feather-nodebot/wiki). I'm working on making a more generic version, as this was for a specific event and has elements (competition schedule, connection to a bot motion-path visualization dashboard, etc.) that aren't broadly applicable.

Tagging @BrianGenisio, as we talked about this issue yesterday and he kindly offered his help.

monteslu, 2017-01-14:
Keep your eyes on this PR:  https://github.com/firmata/firmata.js/pull/149

Soon as it lands we can get to having j5 make use of it to get PWM resolution working.

monteslu, 2017-01-14:
also more info about it in the conversation here that could be helpful:  https://github.com/firmata/firmata.js/pull/144

rwaldron, 2017-03-30:
Hi! Sorry for the delay, I was on parental leave from January 20th until last week and I'm still trying to catch up with everything. I appreciate your patience ❤️

dtex, 2018-01-11:
Hi @hxlnt 

This issue is super important (arguably the most important issue we have open right now) and my closing this certainly doesn't mean that we don't care, we just haven't gotten to it yet. I'm worried about it getting lost in issues so rather than leave it languishing we have created a [Requested Features](../wiki/Requested-Features) page and added your the request for leveraging io.resolution there.

There are two other related issues that I am closing along with this one, so we definitely have mots of people interested in this. We just need a motivated contributor.

closed 2018-01-11 by dtex

dtex, 2018-09-22:
This landed in #1495 and will be in the next release of Johnny-Five (v1.1.0?)