rwaldron/johnny-five #1276: Use IO Plugin .RESOLUTION.* if available

rwaldron, 2017-01-16:
I'm formulating a plan for how to incorporate this: https://github.com/firmata/firmata.js/commit/bb46a890dbe31ac7ae4122f53c24b824dcee17fd across all IO Plugins

hxlnt, 2017-01-23:
Awesome--looks like this will address the issue I opened a few weeks ago (#1275). Thanks!

As I mentioned in that issue, I have a Feather HUZZAH Nodebots/J5 [repo](https://github.com/hxlnt/feather-nodebot) that has J5 motor PWM working with a generic L9110 dual-channel H-bridge board (no FeatherWings) if it's helpful to you for testing purposes. The `index.js` has a bunch of other stuff for dashboard connectivity, so I made a trimmed-down version called `index-withoutdashboard.js` that just has the bot driving code. There's also a Wiki with setup instructions, wiring diagrams, etc. Feel free to tag me or find me on [twitter](http://twitter.com/partytimehxlnt) if you have questions about it or if I can help!

rwaldron, 2017-03-30:
This has shipped in Firmata.js

dtex, 2018-01-11:
Hi @rwaldron 

This issue is super important (arguably the most important issue we have open right now) and my closing this certainly doesn't mean that we don't care, we just haven't gotten to it yet. I'm worried about it getting lost in issues so rather than leave it languishing we have created a [Requested Features](../wiki/Requested-Features) page and added your the request for leveraging io.resolution there.

There are two other related issues that I am closing along with this one, so we definitely have mots of people interested in this. We just need a motivated contributor.

closed 2018-01-11 by dtex

dtex, 2018-09-22:
This landed in #1495 and will be in the next release of Johnny-Five (v1.1.0?)