rwaldron/johnny-five #1277: Using pin A2 as relay control pin

mwittig, 2017-01-20:
I have an arduino board and would like to use pin A2 as the relay control pin, but this does not seem work. It seems like port D2 is used instead. I am wondering why that is? I think, it should be possible to use the analog pins as several tutorials are suggesting. Do I need a customized Firmata configuration for this?

rwaldron, 2017-01-20:
Can you provide the code you're using?

mwittig, 2017-01-20:
Sure. It is pretty basic. I have an LED attached to D2 which goes on and off when the relay is toggled.

```
var five = require('johnny-five');
var board = new five.Board({
    repl: false,
    port: "/dev/ttyUSB1"
  });

board.on('ready', function() {

  try {
    var relay = new five.Relay({
      pin: "A2"
    });

    setInterval(function() {
      relay.toggle();
    }, 5000);
  }
  catch  (error) {
    console.log("Ooops", error);
  }
});

board.on('error', function(error) {
  console.log("Ooops", error);
});
````

rwaldron, 2018-01-05:
@dtex just pointed out that this may be related to #1234

PR #1427 (2018-02-20): Fix relay to use analog output
Fixes #1277
closed 2018-02-20 by rwaldron