rwaldron/johnny-five #1279: can't connect the right Port

salmanfarisvp, 2017-01-22:
my arduino is connected on COM 10 Port , but when i upload the blink code 

```
> 1485113005501 Device(s) COM1,COM10
> 1485113005561 Connected COM1
> 1485113015567 Device or Firmware Error A timeout occurred while connecting to the Board.
```

i got this error , am sure problem is trying to connect port COM1 , but original on is COM10

dtex, 2017-01-22:
I'm not sure why it's seeing a device on COM1, but I know you can specify which port to connect to:

````
var five = require("johnny-five");
var board = new five.Board({ port: "COM10" });

board.on("ready", function() {
  var led = new five.Led(13);
  led.blink(500);
});
````

salmanfarisvp, 2017-01-24:
thank you @dtex 
now it's working fine..

closed 2017-01-24 by salmanfarisvp