rwaldron/johnny-five #1280: Q: PWM resolution with PCA9685

mwittig, 2017-01-23:
Question: Is it possible to use 12-bit PWM for PCA9685 with J5? Looking at the Led and Expander code the PWM output seems to be restricted to 8-bit. Is it planned to support this feature in a future release?

dtex, 2017-01-24:
Yes, this is something we need to add. Right now there is work to add support for boards that support more than 8-bits to Firmata and that will inform what we need to do for Expander (the class that wraps the PCA-9685).

HipsterBrown, 2017-01-24:
There is an open issue around Plugin Resolution here -> https://github.com/rwaldron/johnny-five/issues/1276

mwittig, 2017-01-24:
Excellent, thank you very much for the feedback. Looking at this from the top-level I can imagine the following use cases:
* query the  expander or device (e.g. Led) about native PWM resolution and use the supported range, e.g. via a new pwm() method
* use 8-bit PWM and let the controller scale the value to the native range, e.g. when using Led.brightness if this remains unchanged

dtex, 2018-01-11:
Hi @mwittig , 

This issue is super important (arguably the most important issue we have open right now) and my closing this certainly doesn't mean that we don't care, we just haven't gotten to it yet. I'm worried about it getting lost in issues so rather than leave it languishing we have created a [Requested Features](../wiki/Requested-Features) page and added your the request for leveraging io.resolution there.

There are two other related issues that I am closing along with this one, so we definitely have mots of people interested in this. We just need a motivated contributor.

closed 2018-01-11 by dtex

dtex, 2018-09-22:
This landed in #1495 and will be in the next release of Johnny-Five (v1.1.0?)