rwaldron/johnny-five #1281: PCF8591 does not support DA conversion

CzechUavGuy, 2017-01-27:
I would like to use PCF8591 as a DA converter. I have found that it is already supported as an expander, but it seems that only its AD capabilities are implemented.
Am I missing something?
If it is correct, do you plan to support its DA feature soon? Or should I try implementing it myself?

CzechUavGuy, 2017-01-28:
I have tried to add another pin to expander, with ANALOG OUT capability. I think you don't want to name it this way, but rather PWM? I have absolutely no idea if this is the correct way to implement it, but so far it seems to do what I need. I haven't tested whether I have not broken something, for example ANALOG IN features.
Proposed patch is this:
In file lib/expander.js update section for PCF8591 so that it looks like this:
```
  PCF8591: {
    ADDRESSES: {
      value: [0x48]
    },
    REGISTER: {},
    initialize: {
      value: function(opts) {
        var state = priv.get(this);

        state.control = 0x45;
        state.reading = false;

        this.address = opts.address || this.ADDRESSES[0];

        opts.address = this.address;
        this.io.i2cConfig(opts);

        Object.assign(this.MODES, this.io.MODES);

        for (var i = 0; i < 4; i++) {
          this.pins.push({
            supportedModes: [
              this.MODES.ANALOG
            ],
            mode: 1,
            value: 0,
            report: 0,
            analogChannel: i
          });
        }
	this.pins.push({
	  supportedModes: [
	    this.MODES.PWM
	  ],
	  mode: 0,	//no idea about this line
	  value: 0,	//no idea about this line
	  report: 0,	//no idea about this line
	  analogChannel: 4
	});

        this.analogPins.push(0, 1, 2, 3, 4);

        this.io.i2cWrite(this.address, state.control);

        this.name = "PCF8591";
        this.isReady = true;

        this.emit("connect");
        this.emit("ready");
      }
    },
    normalize: {
      value: function(pin) {
        if (typeof pin === "string" && pin[0] === "A") {
          return +pin.slice(1);
        }
        return pin;
      }
    },
    pinMode: {
      value: function(pin, mode) {
        this.pins[pin].mode = mode;
      }
    },
    analogRead: {
      value: function(pin, callback) {
        var state = priv.get(this);
        var pinIndex = pin;

        this.pins[pinIndex].report = 1;

        this.on("analog-read-" + pin, callback);

        // Since this operation will read all 4 pins,
        // it only needs to be initiated once.
        if (!state.reading) {
          state.reading = true;

          this.io.i2cRead(this.address, 4, function(data) {
            var value;
            for (var i = 0; i < 4; i++) {
              value = data[i] << 2;
              this.pins[i].value = value;

              if (this.pins[i].report) {
                this.emit("analog-read-" + i, value);
              }
            }
          }.bind(this));
        }
      }
    },
    pwmWrite: {
      value: function(pin, value) {
	console.log("writing " + value + " to pin " + pin);
        if (this.pins[pin] === undefined) {   //TODO better error checking?
          throw new RangeError("Invalid PCF8591 pin: " + pin);
        }

        value = Board.constrain(value, 0, 255);
        this.io.i2cWrite(this.address, [0x45, value]);
      }
    }
  },
```

example of usage:

```
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var virtual = new five.Board.Virtual(
    new five.Expander("PCF8591")
  );
  virtual.io.pwmWrite(4, 128);
});
```

CzechUavGuy, 2017-01-30:
Created pull request #1285

fivdi, 2017-02-27:
I'm not sure if it's a good idea to reuse the PWM pin mode to configure pins for PWM or DAC. Maybe Johnny-Five should have a new pin mode for DAC. That being said, introducing a new pin mode is almost certainly an non-trivial task.

closed 2019-01-04 by stale[bot]