rwaldron/johnny-five #1282: SHT31D doesn't change temperature or humidity

aabm00, 2017-01-27:
I am testing SHT31D on YUN

The code is:

```
var five = require('johnny-five');

var board = new five.Board({ port: "/dev/ttyATH0"});

board.on('ready', function() {

  var multi = new five.Multi({
    controller: "SHT31D",
    address: 0x44,
    freq:2000
  });

  console.log('Reading i2c SHT31D');

  //this.repl.inject({ write : write });

  multi.on("data", function() {
    console.log("Thermometer");
    console.log("  celsius           : ", this.thermometer.celsius);
    console.log("  fahrenheit        : ", this.thermometer.fahrenheit);
    console.log("  kelvin            : ", this.thermometer.kelvin);
    console.log("--------------------------------------");

    console.log("Hygrometer");
    console.log("  relative humidity : ", this.hygrometer.relativeHumidity);
    console.log("--------------------------------------");
  });

  console.log('done.');

});
```
But I get always the same lecture it doesn`t matter if I make a breath on the sensor, and the humidity is a negative value.

> --------------------------------------
> Thermometer
>   celsius           :  19.32
>   fahrenheit        :  66.78
>   kelvin            :  292.47
> --------------------------------------
> Hygrometer
>   relative humidity :  -35.56
> --------------------------------------

But when I test the program with **SHT31Test.ino** provided by Adafruit I get different values,
 and the humidity is correct (a positive value);

> Temp *C = 19.89
> Hum. % = 63.10
> 
> Temp *C = 19.88
> Hum. % = 63.16
> 
> Temp *C = 19.91
> Hum. % = 63.07
> 
> Temp *C = 19.88
> Hum. % = 63.09
> 

> Temp *C = 23.24
> Hum. % = 78.21
> 
> Temp *C = 22.57
> Hum. % = 84.17
 
And when I breath on the sensor I get:

> Temp *C = 23.24
> Hum. % = 78.21
> 
> Temp *C = 22.57
> Hum. % = 84.17

fivdi, 2017-01-28:
There's currently a [similar issue](https://github.com/rwaldron/johnny-five/issues/1278) with the HTU21D. I don't have an SHT31D to perform tests. Does it work if the following [two lines of code in imu.js](https://github.com/rwaldron/johnny-five/blob/master/lib/imu.js#L118-L119) are modified from:
```js
          computed.temperature = int16(data[0], data[1]);
          computed.humidity = int16(data[3], data[4]);
```
to:
```js
          computed.temperature = uint16(data[0], data[1]);
          computed.humidity = uint16(data[3], data[4]);
```
?

balda, 2017-01-30:
I have same issue and find a "fix": https://github.com/rwaldron/johnny-five/issues/1253

But waiting between i2c calls is maybe not a good solution ...

balda, 2017-01-30:
Here is my code change: https://github.com/balda/johnny-five/commit/8c704b367a5953a09df29b76e4c1e9927f0b87b2

aabm00, 2017-01-30:
Hi, thank you both.

@fivdi changing int16 by uint16 doesn't work for SHT31D. 
@balda your modification works fine. It will be nice that @rwaldron included it to the download version.

Thanks again!!!

balda, 2017-02-01:
@fivdi i confirm that your fix works: i've encountered the problem when RH >= 50% (value "flip" to negative). With `uint16`, it works perfectly :-)

@aabm00 i have updated my branch with @fivdi fix.

fivdi, 2017-02-01:
I wonder if the int16/uint16 issue is a general problem in imu.js?

aabm00, 2017-02-02:
Ok, So the finally solution is a combination of you both @balda to change the values over time and @fivdi for the negative problem.

Thanks !!! :)

aabm00, 2017-02-06:
Hi.

I have been testing the SHT31D for different intervals between data, freq = 250 and freq = 500 and it works fine, but after about one hour the readings get stuck and do not change any more.

@balda have you notice the same thing or it's working fine for you?

Now I am testing for frequency of 1000 ms.

Thanks

balda, 2017-02-06:
@aabm00 i have 2 SHT31D running (plugged to an Arduino Micro connected to a Raspberry Pi 3). First is running since 08 Dec 2016, second since 31 Jan 2017 (with @fivdi fix).
And i haven't noticed problem: but my frequency is 10000 ms.

I'll try with shorter frequency (but i don't know when ;) )

@fivdi i received a `Si7021` (multi temp/humidity). When i'll have time to test it, i'll check if i encountered same int16/uint16 issue.

aabm00, 2017-02-06:
The SHT31D has been working for 2 hours until now and it works fine so it look that for frequencies less that 1000 ms it fails after an hour.
If the results change I notify it

balda, 2017-02-08:
Thanks @aabm00.

@fivdi i tested a `Si7021` and i confirm same bug (jump from 56% to -68% instead of 57%). With `uint16()` fix, it works perfectly. `SI7021` use same driver than `SI7020`(https://github.com/rwaldron/johnny-five/blob/master/lib/imu.js#L1952)

@rwaldron can we propose a PR to solve this bug?

lyzadanger, 2017-02-13:
@balda If you make sure to add appropriate tests, I think a PR would be greatly appreciated!

aabm00, 2017-02-17:
Hi 

The SHT31D has been working for 3 hours with a frequency of 2000 ms and with the imu.js modified from @balda, and it have failed. The readings doesn't change.

aabm00, 2017-02-20:
The same result with interval of 10000 ms, after about 2 hours blocks and don't change the readings.

Some idea from where can come this behaviour

Thanks

balda, 2017-03-06:
@aabm00 i have updated my frequency (down to 1000 ms). I'll tell you soon if i have a bug :)

@lyzadanger thanks! i'll work on tests before submitting a PR.

aabm00, 2017-03-06:
Thanks @balda !!

balda, 2017-03-06:
@aabm00 4h working without problem at 1000ms.

aabm00, 2017-03-07:
Thanks @balda.

If it's working for you, probably i have something going wrong. Because i couldn't find out what was the problem and I can't spend more time at this time I am using the adafruit sketch for now.
If i find out the problem I'll post it.

rwaldron, 2017-03-30:
Hi! Sorry for the delay, I was on parental leave from January 20th until March 20th and I'm still trying to catch up with everything. I'm also sorry that we weren't able to help you resolve this issue :( 

I'm going to close this for now, but if anyone wants to take this on, post here and we will re-open.

closed 2017-03-30 by rwaldron