rwaldron/johnny-five #1287: I2C MasterReader - SlaveWriter - How to control the interval between readings

aabm00, 2017-02-01:
Hi

I am connecting via I2C an Arduino YUN as a master reader and a Arduino One as slave writer.
The YUN has installed the StandardFirmataYun in the ATmega32u4 and Johnny-Five in the Atheros AR9331, The ONE has an arduino sketch.

The goal is that the ONE has installed different analogic devices like thermistors,  and I want to read in the YUN  the data of this thermistors every 5 seconds. But in Johnny-five in the method:

**this.io.i2cRead(address,` bytesToRead, handler)**

There is no a parameter as "freq" to determine what is the interval between readings.
In the other way in  the method **i2cConfig** I added the **delay** parameter:

**this.io.i2cConfig({delay : 2000000});** but it has no effect.

I finally got the result but not from Johnny-five but from the ONE sketch, but I don't know if it is the best way. The code is:

 ==== YUN ==== MASTER READER

 ```
var five = require('johnny-five'),
       _    = require('lodash');

var board = new five.Board({ port: "/dev/ttyATH0"});

board.on('ready', function() {

	var self = this;

  var read =  function(){
    //self.io.i2cRead(address, bytesToRead, handler)
    self.io.i2cRead(8, 27, function(ArrayOfBytes){

      //console.log(ArrayOfBytes);

      var length = ArrayOfBytes.length;
      var message = "";

      for (var i = 0; i < length; i++){
        var code = ArrayOfBytes[i]
        var char = String.fromCharCode(code);
        if (code != 255){
          message = message.concat(char);
        }
        
      }
      if (message.length != 1){
        console.log(message.length, message);
      }

    });
  }

  console.log('Reading i2c...');

  this.io.i2cConfig({delay : 2000000});
  //this.repl.inject({ write : write });

  read();

  console.log('done.');

});
```
 ==== ONE==== SLAVE WRITER

```
#include <Wire.h>

// the value of the 'other' resistor
#define SERIESRESISTOR 1000000    
 
// What pin to connect the sensor to
#define THERMISTORPIN A1 

float reading;
float ohmios;

void setup() {
  Wire.begin(8);                
  Wire.onRequest(requestEvent);
  Serial.begin(9600);
}

void loop() {
  reading = analogRead(THERMISTORPIN);
 
  Serial.print("Analog reading "); 
  Serial.println(reading);
 
  // convert the value to resistance
  ohmios = (1023 / reading)  - 1;
  ohmios = SERIESRESISTOR / ohmios;
  Serial.print("Thermistor resistance "); 
  Serial.println(ohmios);
  Serial.println("");
 
  delay(2000);
}

// function that executes whenever data is requested by master
// this function is registered as an event, see setup()
void requestEvent() {
  if(ohmios != 0){
    char buffer[9];
    dtostrf(ohmios, 5, 2, buffer);
    Wire.write(buffer); 

    ohmios = 0;
  }
}
```

With this implementation every 2 seconds the data in **ohmios** variable is sent, but actually the data is sent constantly but with value **nul** and I control that this data isn't showed in the YUN by the code:

```
      if (message.length != 1){
        console.log(message.length, message);
      }
```
If I don't add this conditional I get print 
1 "\u0000"
1 "\u0000"
....
1 "\u0000"
1 "\u0000"
**7 "9871.67"**
1 "\u0000"
1 "\u0000"
....
1 "\u0000"

Is there another more elegant way to achive it?

Thanks

fivdi, 2017-02-06:
@aabm00 Maybe https://github.com/rwaldron/io-plugins/issues/11#issuecomment-277818700 helps.

fivdi, 2017-02-11:
@aabm00 Can this issue be closed? If so, please hit the `Close issue` button at the bottom of the page :). If I'm not mistaken two ways to resolve the issue were discussed at https://github.com/rwaldron/io-plugins/issues/11

aabm00, 2017-02-13:
Yes, thanks

closed 2017-02-13 by aabm00