rwaldron/johnny-five #1288: SR04 repeating output after a while

glowlabs, 2017-02-03:
I have 3 SR04 sensors connected. Each are setup like this:

centerSonar = new five.Proximity({
        controller: "HCSR04",
        pin: 22,
        freq: 300
    });

After a few minutes one of the sensors will start reporting the same distance over and over. The sensor that gets stuck changes between the 3 so I don't think its a hardware issue.

Also, if I disconnect power to the sensor that is stuck it still outputs the same measurement. 

I am using the FirmataPing.

Any ideas?

lyzadanger, 2017-02-13:
Hi, @glowlabs 

> I have 3 SR04 sensors connected.

Connected to what, if I might ask? What hardware are you using?

rwaldron, 2017-03-30:
The unavoidable process blocking required by these sensors will inevitably result in perceptible latency, or worse. There are many alternative proximity sensors that are supported, as well as an [alternative](http://johnny-five.io/examples/proximity-hcsr04-i2c/) means to using the SR04 without PingFirmata and the burden of process blocking the main uC.

closed 2017-03-30 by rwaldron