rwaldron/johnny-five #1289: five.Buttons is triggering all buttons

willmendesneto, 2017-02-11:
I'm using johnny-five buttons, but testing the code based on documentation example I realise that I can't manage only one buttons per time. The current `buttons` instance is triggering the events in all buttons.

## About the setup
- Arduino UNO;
- MacBook Pro (Retina, 15-inch, Mid 2015);
- NodeJS 5.3.0;
- NPM 3.3.12;
- Johnny-Five 0.10.6;

## About the code 
The used code is:
```javascript
var five = require("johnny-five"),
  board;

board = new five.Board();

board.on("ready", function() {
  var buttons = new five.Buttons({
    pins: [2, 4],
    invert: true,
  });

  buttons.on("press", function(button) {
    console.log("Pressed: ", button.pin);
  });

  buttons.on("release", function(button) {
    console.log("Released: ", button.pin);
  });
});
```

And the output is

[image]

@rwaldron, that's the correct approach or in this scenario we should change the approach and add this in Johnny-Five documentation?

fivdi, 2017-02-11:
If I understand the requirement correctly, if button-n is pressed and a "press" event is emitted for button-n, then no events should be emitted for any other button before button-n is released and it's "release" event is emitted. Is this correct?

willmendesneto, 2017-03-05:
Not. The main goal with that is share if the button is pressed really or not. How we are using events to handle that, every single button will receive this event in broadcast, but I need to know which button is exactly pressed in that moment. 

```javascript

board.on("ready", function() {
  var buttons = new five.Buttons({
    pins: [2, 4],
    invert: true,
  });

  buttons.on("press", function(button) {
    // how to know which button exactly was pressed? 
    console.log("Pressed: ", button.pin);
  });

  buttons.on("release", function(button) {
    // how to know which button exactly was released?
    console.log("Released: ", button.pin);
  });
});
```

dtex, 2017-03-05:
I'm away from my hardware at the moment, but have you tried something like this:

````
 buttons[0].on("press", function(button) {
    // how to know which button exactly was pressed? 
    console.log("The first button was pressed");
  });

 buttons[1].on("press", function(button) {
    // how to know which button exactly was pressed? 
    console.log("The second button was pressed");
  });
````

willmendesneto, 2017-03-05:
Yes @dtex and how we are talking about the same event, both are triggered. I already compare the information inside button, but for now, I can't find anything saying that it's activate or not.

dtex, 2017-03-06:
If you want to manage them separately couldn't you just use two separate button instances:

````
var five = require("johnny-five"),
  board;

board = new five.Board();

board.on("ready", function() {
  var leftButton = new five.Button({
    pins: 2,
    invert: true,
  });

var rightButton = new five.Button({
    pins: 4,
    invert: true,
  });

  leftButton.on("press", function(button) {
    console.log("Left button pressed");
  });

  rightButton.on("press", function(button) {
    console.log("Right button pressed");
  });
});
````

soundanalogous, 2017-03-06:
It sounds like @willmendesneto is saying that when he presses a single button, all buttons fire an event. This sounds like either an issue in the five.Buttons module, or an issue in the user's circuit. Will, are you using internal pull-up resistors or external pull-up or pull-down resistors?

willmendesneto, 2017-03-06:
@dtex I tried that one before use this approach and I had the same issue as well. As I said, the buttons are being triggered using the same event, which means that it doesn't matter if I use `five.Buttons` to handle with all buttons or create an instance per each button using `five.Button`, both didn't work.

willmendesneto, 2017-03-06:
> It sounds like @willmendesneto is saying that when he presses a single button, all buttons fire an event. This sounds like either an issue in the five.Buttons module, or an issue in the user's circuit. 

@soundanalogous It's exactly that. I guess it's something related to Johnny-Five really since I'm using the same buttons and resistors as [fritzing diagram added in johnny-five docs](https://github.com/rwaldron/johnny-five/blob/master/docs/breadboard/button.fzz). If we can send in the event some parameter showing that the button is active or not (or even inside button instance) should solve this issue.

fivdi, 2017-03-07:
@willmendesneto I just tried this with an Arduino UNO and there were no issues. Everything worked as expected. Both buttons were connected as shown [here](https://github.com/rwaldron/johnny-five/blob/master/docs/breadboard/button.png) which is the same as the fritzing you link to above. One change was however made to the program:
```js
var five = require("johnny-five"),
  board;

board = new five.Board();

board.on("ready", function() {
  var buttons = new five.Buttons({
    pins: [2, 4],
//    invert: true,
  });

  buttons.on("press", function(button) {
    console.log("Pressed: ", button.pin);
  });

  buttons.on("release", function(button) {
    console.log("Released: ", button.pin);
  });
});
```
Note that `invert: true` was commented out. It doesn't make sense to use `invert: true` if the buttons are connected to the Arduino as showh [here](https://github.com/rwaldron/johnny-five/blob/master/docs/breadboard/button.png) as it will result in unauthentic `press` events being emitted when the program starts without the buttons actually being pressed. If `invert: true` is used, the output of the program at startup will look something like this although the buttons were not pressed:
```
1488900612858 Device(s) /dev/ttyACM0  
1488900612881 Connected /dev/ttyACM0  
1488900616577 Repl Initialized  
>> Pressed:  2
Pressed:  4
```
If removing `invert: true` doesn't resolve the issue for you please post some photos of the circuit. This will help to figure out what the problem is.

fivdi, 2017-04-01:
@willmendesneto any news on this one?

rwaldron, 2017-04-03:
Before any more software trouble shooting can be done, we need to see a clear picture of the actual button circuit in use. I've just now wired up two buttons and ran your exact code and it worked exactly as it should—so I suspect there is something amiss in the circuit.

willmendesneto, 2017-04-04:
@fivdi @rwaldron I will take a look and if still not working I'll add specific information about the buttons and resistors. I can send it now, but I will give you some feedback in few hours

willmendesneto, 2017-04-15:
Closing this issue right now. Thanks guys!

closed 2017-04-15 by willmendesneto