rwaldron/johnny-five #1290: Grove Temperature and Humidity Sensor Pro

BartNetJS, 2017-02-11:
I'm trying to get the http://wiki.seeed.cc/Grove-Temperature_and_Humidity_Sensor_Pro/ working with johnny-five

As far I can see in there code for arduino [on line 79](https://github.com/Seeed-Studio/Grove_Temperature_And_Humidity_Sensor/blob/master/DHT.cpp) they read 40 bits via the digital pin and convert those bits to temperature and humidity

Any tips how to get started with johnny-five?

fivdi, 2017-02-11:
The Grove Temperature and Humidity Sensor Pro uses a DHT22 sensor. The usage of the DHT22 is documented [here](http://johnny-five.io/examples/multi-DHT22_I2C_NANO_BACKPACK/). Note that a backpack is required. The backpack is the Arduino Nano in the documentation linked to above.

rwaldron, 2017-03-30:
Thanks @fivdi!

closed 2017-03-30 by rwaldron