rwaldron/johnny-five #1292: YUN with J5 v0.9.62 works - with J5 v0.10.6 doesn't

aabm00, 2017-02-17:
Hi 

With J5 v0.10.6 I get the error "error: Forever detected script exited with code: 0." but with the version 0.9.62 everything works fine.
 
In the YUn I have installed:

node 0.10.33-1 
node-serialport 1.4.6-2  

This are the versions that the OpenWRT via opkg install, they can not be upgraded.

Is there some change that can be the reason that the J5 v0.10.6 library doesn't run in the Arduino YUN?

Thanks

rwaldron, 2017-03-30:
That's a really old version of node-serialport and Johnny-Five 0.10.x is set to always use the latest. @reconbot have you encountered any issues with other node-serialport users and Yun?

reconbot, 2017-03-31:
I have not heard anything special - That version of serialport is known to have memory leaks and cause crashes.

rwaldron, 2017-03-31:
@reconbot thanks :)

aabm00, 2017-04-04:
Hi !! Thanks @rwaldron @reconbot 

So what would be the solution?
Node and node-serialport has to be installed with the package system **opkg** of **OpenWRT** in the YUN and the last versions it supports are the ones I mentioned.

node 0.10.33-1
node-serialport 1.4.6-2

So what version of **node-serialport** without problems of memory leaks can I install?
Are all the new versions of J5 compatible with old node versions like the 0.10.x?

Thanks

reconbot, 2017-04-04:
Serialport 4x which works best on node 4 or 6 but should still run on node
0.10

aabm00, 2017-04-04:
Hi @reconbot

The weight of the Serialport  version installed by the opkg of OpenWRT is about 5,97 MB The libraries under node_modules are:
async, bindings, nan, node-pre-gyp, optimistic and sf.

The source of the Serialport 4.0.7 after execute the npm install is about 76,9 MB. 
There is some way to reduce the weight. I supose I can eliminate libraries an only left the same libraries as the wersion 1.4.6 but the  optimistic and sf doesn't exist in the  v 4.0.7

reconbot, 2017-04-04:
Most of the weight comes from node-pre-gyp which you can remove as it's
only used during install time on other platforms.

aabm00, 2017-04-04:
After npm uninstall node-pre-gyp still 74,0 MB

reconbot, 2017-04-04:
I think npm prune will remove all it's dependencies

aabm00, 2017-04-04:
I have tried to delete the 1.4.6 version of Serialport installed by opkg and copy the 4.0.7 but it doesn't work 
I get this error:
[image]

I don't know if OPenWRT coming with windoes has an special version of the serialPort or simply can't manage versions bigger than the 1.4.6

reconbot, 2017-04-04:
You probably need to rebuild the binary.

---
Francis Gulotta
wizard@roborooter.com

aabm00, 2017-04-04:
What I have done is:
1- Download the source of serialPort 4.0.7
2- Unpackt
3- execute npm install inside of node-serialport-4.0.7
4- Rename the folder from node-serialport-4.0.7 to serialport (to match the installed by opkg)
5- Delete the old version from YUN and install the new one.

This is Ok ?

reconbot, 2017-04-04:
Did you do this on the yun or on your computer?

---
Francis Gulotta
wizard@roborooter.com

aabm00, 2017-04-04:
I did it in my computer an copy it in the YUN with WinSCP.  I have an  8GB microSD that extends the disk space of the YUN.

reconbot, 2017-04-06:
Rick, do you know what architecture the YUN has?

Aabm00, You'll need to have the right binaries for the Yun, if you
installed it on your PC and copied it over you'll have the windows binaries
and those wont work on the YUN.

---
Francis Gulotta
wizard@roborooter.com

aabm00, 2017-04-06:
Hi
In this [link ](https://www.arduino.cc/en/Main/ArduinoBoardYun) says:
The Arduino Yún is a microcontroller board based on the ATmega32u4 and the Atheros AR9331. The Atheros processor supports a Linux distribution based on OpenWrt named Linino OS. 

I do the same process described in my last post with all the libraries under **mnt/sda/project/node_modules**  I mean I develop the project initially in my laptop with Windows and install all the libraries with **npm install** and it was fine.

Because of your comments I understand that node-serialport has to be compiled (npm install) with linux and then copied into the YUN.

I don't have Linux, so what can I do, can I download a compiled linux version of the node-serialport 4.0.7?

Thanks

reconbot, 2017-04-06:
It has to be a Linux with the same cpu architecture as the yun.

aabm00, 2017-04-07:
Hi

After a lot of hours the node-serialport 4.0.7 was compiled directly in the YUN.  I unpackt and copied the source  in the YUN and executed npm install
The result is here:

   
```
 login as: root
root@192.168.1.130's password:

BusyBox v1.19.4 (2014-11-13 19:03:47 CET) built-in shell (ash)
Enter 'help' for a list of built-in commands.

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------

root@Arduino:~# cd /usr/lib/node_modules/serialport
root@Arduino:/usr/lib/node_modules/serialport# npm install
npm WARN engine chai-subset@1.5.0: wanted: {"node":">=4"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine eslint@3.19.0: wanted: {"node":">=4"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine request@2.79.0: wanted: {"node":">= 4"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine request@2.81.0: wanted: {"node":">= 4"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine deep-extend@0.4.1: wanted: {"node":">=0.12.0","iojs":">=1.0.0"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine form-data@2.1.2: wanted: {"node":">= 0.12"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine form-data@2.1.2: wanted: {"node":">= 0.12"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine har-validator@4.2.1: wanted: {"node":">=4"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine har-schema@1.0.5: wanted: {"node":">=4"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN deprecated to-iso-string@0.0.2: to-iso-string has been deprecated, use @segment/to-iso-string instead.
npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install the latest version of pug instead of jade
npm WARN deprecated minimatch@0.3.0: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN engine cryptiles@2.0.5: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine boom@2.10.1: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine hoek@2.16.3: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine hoek@2.16.3: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine cryptiles@2.0.5: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine boom@2.10.1: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine string-width@2.0.0: wanted: {"node":">=4"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine esprima@3.1.3: wanted: {"node":">=4"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine is-fullwidth-code-point@2.0.0: wanted: {"node":">=4"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine strip-bom@3.0.0: wanted: {"node":">=4"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine shelljs@0.7.7: wanted: {"node":">=0.11.0","iojs":"*"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine co@4.6.0: wanted: {"iojs":">= 1.0.0","node":">= 0.12.0"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine string-width@2.0.0: wanted: {"node":">=4"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine esprima@3.1.3: wanted: {"node":">=4"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine is-fullwidth-code-point@2.0.0: wanted: {"node":">=4"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine co@4.6.0: wanted: {"iojs":">= 1.0.0","node":">= 0.12.0"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN engine co@4.6.0: wanted: {"iojs":">= 1.0.0","node":">= 0.12.0"} (current: {"node":"0.10.33","npm":"1.4.28"})
npm WARN cannot run in wd serialport@4.0.7 node-pre-gyp install --fallback-to-build (wd=/usr/lib/node_modules/serialport)
bindings@1.2.1 node_modules/bindings

chai-subset@1.5.0 node_modules/chai-subset

sinon-chai@2.9.0 node_modules/sinon-chai

eslint-plugin-promise@1.3.2 node_modules/eslint-plugin-promise

eslint-plugin-standard@1.3.3 node_modules/eslint-plugin-standard

commander@2.9.0 node_modules/commander
└── graceful-readlink@1.0.1

sandboxed-module@2.0.3 node_modules/sandboxed-module
├── stack-trace@0.0.9
└── require-like@0.1.2

lie@3.1.1 node_modules/lie
[... 118 lines cut ...]

nyc@6.6.1 node_modules/nyc
root@Arduino:/usr/lib/node_modules/serialport#
```

But it doesn't work when I run the project in the YUN get this message:

```
root@Arduino:~# /mnt/sd/arduino/iotfs_yun/node_modules/forever/bin/forever /mnt/
sd/arduino/iotfs_yun/iotfs-server.js
warn:    --minUptime not set. Defaulting to: 1000ms
warn:    --spinSleepTime not set. Your script will exit if it does not stay up for at least 1000ms
IOTFS_SERVER YUN SOCKET CONNECTED
It looks like serialport didn't compile properly. This is a common problem and its fix is well documented here https://github.com/voodootikigod/node-serialport#to-install

/mnt/sda1/arduino/iotfs_yun/node_modules/firmata/lib/com.js:50
    throw "Missing serialport dependency";
    ^
Missing serialport dependency
error: Forever detected script exited with code: 8
error: Script restart attempt #1
IOTFS_SERVER YUN SOCKET CONNECTED
^C
root@Arduino:~#
```

reconbot, 2017-04-07:
I can't tell exhausted what's going on but you'll need at least npm 3 (npm
install -g npm) you can also save some space by doing npm install
--production so Dev dependencies don't get installed.

aabm00, 2017-04-07:
But if I install  the version 3 of npm , this version will be compatible with the node 0.10.33-1 installed?

reconbot, 2017-04-07:
If it works on 0.10 at all everything it installs will work.

aabm00, 2017-04-07:
I just installed npm 4.4.4 and I got:

root@Arduino:/usr/lib/node_modules/serialport# npm install -g npm
npm WARN npm npm does not support Node.js v0.10.33
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm You can find the latest version at https://nodejs.org/

reconbot, 2017-04-08:
Did it work?

aabm00, 2017-04-10:
Hi

It didn't work with npm 4.4.4.
It's a tedious work install libraries directly via npm in arduino YUN because is super slow and if the conexion fails (wifi) you have to start again or wrost, you can't have access to the extension micro

aabm00, 2017-04-12:
Hi

I installed in the YUN:

>   npm install –g npm@3.10.9

Then I copied the source flles of the **node-serialport 4.0.7** from [github](https://github.com/EmergingTechnologyAdvisors/node-serialport/releases) in the YUN under **/usr/lib/node_modules/serialport** and executed **npm install --production** 

> root@Arduino:/usr/lib/node_modules/serialport# npm install --production

Once installed I run my program in YUN an get the same error:

    root@Arduino:~# /mnt/sd/arduino/iotfs_yun/node_modules/forever/bin/forever /mnt/
    sd/arduino/iotfs_yun/iotfs-server.js
    warn:    --minUptime not set. Defaulting to: 1000ms
    warn:    --spinSleepTime not set. Your script will exit if it does not stay up for at least 1000ms

    /mnt/sda1/arduino/iotfs_yun/node_modules/firmata/lib/com.js:50
        throw "Missing serialport dependency";
        ^
    Missing serialport dependency
    It looks like serialport didn't compile properly. This is a common problem and its fix is well 
    documented here https://github.com/voodootikigod/node-serialport#to-install

The only way it works is installing node-serialport with **opkg** :

    root@Arduino:~# opkg install node-serialport
    Installing node-serialport (1.4.6-2) to root...
    Downloading http://downloads.arduino.cc/openwrtyun/1/packages/node-serialport_1.4.6-2_ar71xx.ipk.
    Configuring node-serialport.

I guess I'm doing something wrong, but I do not know what.

I would like to know what Linux version could install on my PC to be able to perform the compilation in PC, because doing it in YUN is tedious.
What do I need to know? You spoke of architecture but I do not know exactly what you mean

Thanks again

closed 2018-08-28 by stale[bot]