rwaldron/johnny-five #1294: Documentation: Multi and Options

lyzadanger, 2017-02-24:
Hi!

[Multi API docs](http://johnny-five.io/api/multi/) don't currently document that the constructor will pass on options to the constituent components (e.g. `freq`), which is a handy feature. Probably a few other nuances that could be added to that docs page as well (distinction of individual vs. aggregate events on each component), and a vetting that docs contain what needs containing...

I'd be happy to take this on if it's something that is useful to do.

lyzadanger, 2018-01-14:
Hi! Got a chance to update the Wiki:

https://github.com/rwaldron/johnny-five/wiki/Multi

dtex, 2018-01-14:
This looks good to me. I know ```opts.freq``` is on the hit list for a breaking change but that seems beside the point for now.

Thank you @lyzadanger !

closed 2018-01-14 by dtex