rwaldron/johnny-five #1295: An LCD displays garbage if an IO plugin is fast at digital output

fivdi, 2017-02-26:
A parallel LCD displays garbage if the IO plugin being used can perform fast digital output.

The execution times for the various LCD instructions are specified on pages 24 and 25 of the [HD44780 datasheet](https://www.sparkfun.com/datasheets/LCD/HD44780.pdf). If the execution time for an instruction is a small number of milliseconds, Johnny-Five takes this time into account. An example of such an instruction is the "return home" instruction. As can be seen in the code [here](https://github.com/rwaldron/johnny-five/blob/v0.10.4/lib/lcd.js#L843-L847), there is a 2 millisecond delay after sending the home instruction to the LCD. Waiting for 2 milliseconds gives the LCD time to complete the operation before the next instruction is sent to the LCD.

However, most instructions have an execution time of 37 microseconds that Johnny-Five doesn't take into account. While this may not have been an issue for most IO plugins in the past, it appears to be an issue for [pi-io](https://github.com/fivdi/pi-io). pi-io is fairly fast at digital output and can perform almost 2.000.000 digital writes per second which means that a single digital write can be performed in approximately 500 nanoseconds. Because Johnny-Five isn't waiting for the instructions with an execution time of 37 microseconds to complete, it's relatively easy for pi-io to begin executing an instruction before the previous instruction has completed.

The issue can be reproduced with [lcd-stress-test.js](https://github.com/fivdi/pi-io/blob/v0.1.1/test/lcd-stress-test.js) on a Raspberry Pi 3. When run, this program will display garbage on the LCD immediately.

It looks like the issue can be resolved by modifying [this code](https://github.com/rwaldron/johnny-five/blob/v0.10.4/lib/lcd.js#L755-L775) from:
```js
LCD.prototype.send = function(value) {
  var pin = 0;
  var mask = {
    4: 8,
    8: 128
  }[this.bitMode];

  for (; mask > 0; mask = mask >> 1) {
    this.io.digitalWrite(
      this.pins.data[pin],
      this.io[value & mask ? "HIGH" : "LOW"]
    );
    pin++;
  }

  ["LOW", "HIGH", "LOW"].forEach(function(val) {
    this.io.digitalWrite(this.pins.en, this.io[val]);
  }, this);

  return this;
};
```
to:
```js
// Waits for the specified number of microseconds.
function sleepus(usDelay) {
  var startTime = process.hrtime(),
    deltaTime,
    usWaited = 0;

  while (usDelay > usWaited) {
    deltaTime = process.hrtime(startTime);
    usWaited = (deltaTime[0] * 1E9 + deltaTime[1]) / 1000;
  }
}

LCD.prototype.send = function(value) {
  var pin = 0;
  var mask = {
    4: 8,
    8: 128
  }[this.bitMode];

  for (; mask > 0; mask = mask >> 1) {
    this.io.digitalWrite(
      this.pins.data[pin],
      this.io[value & mask ? "HIGH" : "LOW"]
    );
    pin++;
  }

  ["LOW", "HIGH", "LOW"].forEach(function(val) {
    this.io.digitalWrite(this.pins.en, this.io[val]);
  }, this);

  sleepus(37); // <<<<<<<<<<========== NEW NEW NEW

  return this;
};
```
Note that the following snippet of code from above is also likely to become critical at some stage:
```js
  ["LOW", "HIGH", "LOW"].forEach(function(val) {
    this.io.digitalWrite(this.pins.en, this.io[val]);
  }, this);
```
This code generates a short high level pulse on the enable pin. If pi-io is being used, the high level pulse width will be approximately 500 nanoseconds. At VCC = 3.3V, the minimum enable pulse width is specified as 450 nanoseconds on page 49 of  the [HD44780 datasheet](https://www.sparkfun.com/datasheets/LCD/HD44780.pdf).

fivdi, 2017-02-26:
The code above that waits 37 microseconds for LCD instructions to complete worked for a while but not for long enough. I'm not sure exactly how long it worked as I was away from the LCD.

An additional modification which changes the following code from:
```js
  ["LOW", "HIGH", "LOW"].forEach(function(val) {
    this.io.digitalWrite(this.pins.en, this.io[val]);
  }, this);
```
to:
```js
  this.io.digitalWrite(this.pins.en, this.io.LOW);
  this.io.digitalWrite(this.pins.en, this.io.HIGH);
  sleepus(1);
  this.io.digitalWrite(this.pins.en, this.io.LOW);
```
gives far better better results. With both modifications [lcd-stress-test.js](https://github.com/fivdi/pi-io/blob/v0.1.1/test/lcd-stress-test.js) has been running successfully for about five hours now.

fivdi, 2017-02-27:
The test ran for an additional eight hours over night without issue. The `sleepus(1)` and `sleepus(37)` calls added to the code above are meaningless when Johnny-Five is running on a host computer connected to an Arduino UNO via a serial port. Not totally meaningless, they waste CPU cycles, but that's about it.

To be honest, I don't actually understand the purpose of `sleep()` calls when an Arduino UNO is being used. For example, consider [this code](https://github.com/rwaldron/johnny-five/blob/v0.10.4/lib/lcd.js#L843-L847) which returns the cursor to the home position:
```js
LCD.prototype.home = function() {
  this.command(this.REGISTER.HOME);
  sleep(2);
  return this;
};
```
If I'm not mistaken calling `this.command(this.REGISTER.HOME)` will result in fourteen calls to `this.io.digitalWrite` in [send](https://github.com/rwaldron/johnny-five/blob/v0.10.4/lib/lcd.js#L740-L775). This, in turn, will result in fourteen calls to serialport to send the appropriate information to the Arduino UNO. These calls to serialport are write and forget calls that are executed asynchronously and in parallel to the code in the JavaScript thread. The `sleep(2)` call in the `home` method above is therefore highly likely to start sleeping before the first byte has been sent to the Arduino UNO. It's also highly likely to have stopped sleeping before the last byte has been sent. Or am I totally misunderstanding things here?

rwaldron, 2017-03-30:
@fivdi would you mind writing a patch for these corrections?

fivdi, 2017-03-30:
> @fivdi would you mind writing a patch for these corrections?

@rwaldron ok, I'll write a patch.

fivdi, 2017-05-07:
Fixed by #1338

PR #1338 (2017-05-09): LCD: Wait for fast instructions to complete
Fixes #1295
closed 2017-05-09 by rwaldron