rwaldron/johnny-five #1297: five.Thermometer sets an unclearable interval when constructed

islemaster, 2017-03-02:
When constructed, `five.Thermometer` [calls `setInterval()` with the initial sampling frequency](https://github.com/rwaldron/johnny-five/blob/7ec4648249a73a90776e90fd0f13f71acbac35a0/lib/thermometer.js#L985-L1001) but does not track the interval key, and thus does not expose any way to pause or stop the component from emitting `'data'` events.

It would be nice to expose `enable()` and `disable()` methods [like `five.Sensor` does](https://github.com/rwaldron/johnny-five/blob/master/lib/sensor.js#L303-L319), even if we want the Thermometer component to start reading immediately.  This would allow the component to be cleaned up and removed if necessary.

rwaldron, 2017-03-30:
Hi! Sorry for the delay, I was on parental leave from January 20th until March 20th and I'm still trying to catch up with everything. I appreciate your patience ❤️

islemaster, 2017-03-30:
No problem!  We're making steady progress on our end - I figured I'd get back to submitting a PR for this myself soon-ish anyway.  Thanks! (And congratulations?)

islemaster, 2017-05-26:
I've proposed a fix for this.  Please take a look!

islemaster, 2017-06-02:
This is resolved as of [v0.11.0](https://github.com/rwaldron/johnny-five/releases/tag/v0.11.0) by the changes in #1346.  Thanks!

closed 2017-06-02 by islemaster