rwaldron/johnny-five #1301: Tessel joystick doesn't work

jdnichollsc, 2017-03-08:
Hi guys!

I'm using the following code but the **change** event doesn't work => https://github.com/jdnichollsc/Tessel/blob/master/example3/index.js 

Any suggestion?

dtex, 2017-03-08:
Which joystick are you using, and can we see a pic of your wiring?

jdnichollsc, 2017-03-08:
Hi man, thanks for your quickly response!

Check the pic please 
[image]

Thanks, Nicholls

dtex, 2017-03-08:
Cool. A few of more questions:

You are using ```t2 run``` not ```t2 push``` correct?

Where does the "SW" pin on the joystick go?

Who makes this joystick?

jdnichollsc, 2017-03-08:
Thanks!
Yes, I'm using t2 run to test. SW isn't connected and I'm trying to do the following example but using **Tessel 2** => http://johnny-five.io/examples/joystick/

dtex, 2017-03-08:
One thing that comes to mind is the 5v supply might not share a ground with the Tessel port. Can you try using the gnd pin on port B instead of the ground next to the 5v pin?

jdnichollsc, 2017-03-08:
Check the console output:
```js
D:\Electronica\Tessel\Starter\example3>t2 run index.js
INFO Looking for your Tessel...
INFO Connected to Nicholls.
INFO Building project.
INFO Writing project to RAM on Nicholls (986.112 kB)...
INFO Deployed.
INFO Running index.js...
1474488778815 Device(s) Tessel 2 (Nicholls)
1474488778968 Connected Tessel 2 (Nicholls)
1474488779024 Repl Initialized
>> Testing...
Joystick
  x :  0
  y :  0
--------------------------------------
Joystick
  x :  0.005870841443538666
  y :  -0.001956947147846222
--------------------------------------
```

But the change event of the joystick doesn't work

dtex, 2017-03-08:
I missed the confused emoji there. My comment about using the gnd on port B instead of the gnd next to the 5V pin is based on a suspicion that w/o a common baseline on gnd you will not get good analog readings.

Notice in [this diagram](https://cdn.sparkfun.com/assets/learn_tutorials/5/3/7/motor-diagram.png) that they are not using the gnd pin next to 5VIN. Everything is grounded on Port A's gnd pin.

jdnichollsc, 2017-03-08:
Hi @dtex,
My confused emoji is because I have the same issue using the gnd on port B :(

rwaldron, 2017-03-08:
Have you tried any other pin combo on port B? That is, not 0 and 1?

jdnichollsc, 2017-03-08:
Hi @rwaldron! Yes but the same result.

boneskull, 2017-03-09:
@jdnichollsc I have a module just like that. Let me try it.

boneskull, 2017-03-09:
@jdnichollsc I can reproduce wonky behavior here.

Basically, the first few readings are OK (though you'd think a joystick at rest wouldn't produce so much data, but it's a piece of garbage anyhow):

```
1489044263452 Device(s) Tessel 2 (bloath)  
1489044263609 Connected Tessel 2 (bloath)  
1489044263666 Repl Initialized  
>> Joystick
  x :  0
  y :  0
--------------------------------------
Joystick
  x :  0.011741682887077332
  y :  -0.011741682887077332
--------------------------------------
Joystick
  x :  0
  y :  -0.013698630034923553
--------------------------------------
Joystick
  x :  0.017612524330615997
  y :  -0.00978473573923111
--------------------------------------
Joystick
  x :  0.011741682887077332
  y :  -0.017612524330615997
--------------------------------------
Joystick
  x :  0.007827788591384888
  y :  0.003913894295692444
--------------------------------------
Joystick
  x :  0
  y :  -0.005870841443538666
--------------------------------------
Joystick
  x :  0
  y :  0.007827788591384888
--------------------------------------
Joystick
  x :  0.003913894295692444
  y :  -0.005870841443538666
--------------------------------------
Joystick
  x :  0.015655577182769775
  y :  -0.007827788591384888
--------------------------------------
Joystick
  x :  0.015655577182769775
  y :  -0.005870841443538666
^C^C
```

...those are my `SIGINT`s, since it froze at that point.

Speculation:  `Joystick` doesn't extend `Sensor`, so I'm not able to set the read frequency, afaik--it's just a firehose.  Applying [throttle](https://lodash.com/docs/4.17.4#throttle) to the callback doesn't seem to make a difference, so I'd start looking somewhere other than `process.stdout`.

Here's my setup, which is over USB:

[image]

What is this joystick?  It's a generic clone of a generic Keyes module.  $0.55, shipped free.

dtex, 2017-03-09:
I've ordered one.

jdnichollsc, 2017-03-21:
Any update guys? :)

dtex, 2017-03-21:
Mine hasn't arrived yet. Seller's tracking app doesn't work so I opened a ticket with the seller to find out what's up.

rwaldron, 2017-03-22:
I have that same Keyes joystick, as well as several other brands, but most are in New York and I'm currently in Boston. On hand I have: 

- Keyes
- Sainsmart

...And they look very similar—probably produced by same manufacturer, but at different times. I'll use the Keyes joystick, with this program: 

```js
var Tessel = require("tessel-io");
var five = require("johnny-five");
var board = new five.Board({
  io: new Tessel()
});

board.on("ready", function() {
  var joystick = new five.Joystick({
    pins: ["B6", "B7"]
  });

  joystick.on("change", function() {
    console.log("Joystick");
    console.log("  x : ", this.x);
    console.log("  y : ", this.y);
    console.log("--------------------------------------");
  });
});
```

Before I dive into my testing, I need to explain something: `t2` will try to connect to your Tessel by USB or Wifi, which it chooses is only determined by which discovery process completes quicker. Deployment is _almost_ identical for both, differing only in the constraints imposed by the method, ie. ssh or usb. Either way, your code is deployed and executed and everything is cool... Unless your program is console.log'ing... If the program was deployed and initiated over USB, it's subject to limited throughput... too much output can overwhelm and subsequently crash the remote process. This behaviour is actually totally ok, because ultimately programs running on a Tessel won't be logging to console anyway (when pushed into flash memory). You can force the deployment to occur over wifi with: `t2 run index.js --lan`. 

All of that aside, I was able to run the program above with the Keyes joystick (actually both, but I don't think that makes a difference) and observe the behavior that I expect: a lot of change events, logging at a uselessly high rate. Try the `--lan` flag and report back. We'll work through this together ;)

dtex, 2017-04-02:
I was able to replicate this today. With just the joystick I had the same experience as Rick, but once I started adding devices it would just freeze up after a few seconds.

I think @boneskull is on to something. It would be a good idea to just wrap a pair of sensors here.

boneskull, 2017-04-02:
Is this "USB bottleneck" unique to the Tessel? I've never had any such issue with opening serial ports on other devices. They all dump serial data with impunity. 

I'm pretty sure I reproduced the same thing with `--lan`. Regardless, the above output I pasted is literally everything. That's not a terribly large amount of data; color me surprised if it causes a buffer problem.

rwaldron, 2017-04-04:
> Is this "USB bottleneck" unique to the Tessel? I've never had any such issue with opening serial ports on other devices. They all dump serial data with impunity.

It's only a "bottleneck" in the sense that USB connections on the Tessel 2 were never intended to be used for anything like this. Programs that are deployed to the Tessel 2's flash won't be logging to a console. 

> That's not a terribly large amount of data; color me surprised if it causes a buffer problem.

Here is the source if you'd like to try and optimize: 

- https://github.com/tessel/t2-firmware/blob/master/firmware/usb.c
- https://github.com/tessel/t2-firmware/blob/master/firmware/usbpipe.c
- https://github.com/tessel/t2-firmware/blob/master/firmware/usbserial.c
- https://github.com/tessel/t2-firmware/blob/master/boot/usb.c

boneskull, 2017-04-04:
> Programs that are deployed to the Tessel 2's flash won't be logging to a console.

Yes, but neither will my Arduino.  :smile:  Until I'm ready to stuff it in an enclosure and put my firmware "in production", I use `Serial.println()` extensively--sometimes dumping stupid amounts of data.

If we're sure the culprit is the USB driver and the problem *isn't* already known, a new issue should be created in `t2-firmware`.

@jdnichollsc Can you create a new joystick class which extends `Sensor`, and retains the implementation of the current J5 code?  Then, you can reduce the read frequency.  That might help.

Assuming it works, I'm not convinced this workaround needs to be committed to J5.  Besides the obvious semantic disagreement, *any* large+fast amount of output over Tessel 2's USB could cause the same crash.  Pretty sure neither J5 proper nor `tessel-io` can do anything about it?

dtex, 2017-04-21:
@jdnichollsc So, I found a cure on my end. I'm embarrassed to say that a simple ```t2 update``` did the trick. It was an Tessel I hadn't used in a while and was running 0.0.13. I updated to 0.0.16 and the problem went away.

Note that I haven't left it running for a particularly long time, but so far so good.

rwaldron, 2017-04-21:
Ohhhhhhhh. Gah. I should've remembered to offer this suggestion :(

dtex, 2017-04-26:
@jdnichollsc I spent some more time with this and found that it was still freezing (it was just less consistent). @rwaldron made some changes to tessel-io which are available in version 0.11.0. Using a T2 and the same joystick from your picture (as best I can tell) it now works without freezing.

Just ```npm install tessel-io``` 

Please let us know if it works for you.

jdnichollsc, 2017-04-26:
Ohh sorry guys, I changed jobs and I was very lost. Let me check your comments and try here! 
Thanks for the amazing help! 👍

jdnichollsc, 2017-04-28:
Hi @dtex @rwaldron @boneskull 
Works beautiful with the version 0.11.0! Thanks for the amazing help! 💯 
Let me follow on Github! I owe you guys a beer! haha 🍻 

Guys, what's the correct way to connect via LAN?

```
jdnichollsc:example3 jdnichollsc$ t2 provision
INFO Looking for your Tessel...
INFO Connected to Nicholls.
INFO Authenticating Tessel with public key...
INFO Tessel is already authenticated with this computer.
jdnichollsc:example3 jdnichollsc$ t2 run index.js --lan
INFO Looking for your Tessel...
WARN No Authorized Tessels Found.
jdnichollsc:example3 jdnichollsc$
```
Thanks in advance 🕹 👍

dtex, 2017-04-28:
Have you followed the instructions here:

http://tessel.github.io/t2-start/wifi.html

If so, what's it saying after you run t2 wifi ?

rwaldron, 2017-04-28:
@jdnichollsc so glad to hear that this was resolved. 

I'm going to ping @hipsterbrown for additional support on any wifi issues :)

HipsterBrown, 2017-04-28:
:+1: to @dtex pointer to http://tessel.github.io/t2-start/wifi.html

I'll try to help if there are any issues.

dtex, 2017-08-17:
Looks like the original issue was solved so closing.

@jdnichollsc If you are still having issues with the T2 wifi, I'd ask for help over on their [Slack channel](https://tessel.slack.com/messages/C04R290NJ/). You'll find a lot of the same folks that were on this thread as well as a bunch of other Tessel experts who can help.

closed 2017-08-17 by dtex