rwaldron/johnny-five #1303: ADXL345 accelerometer - data event

alex-def, 2017-03-11:
Hello,

Regarding API (http://johnny-five.io/api/accelerometer/#events) , i would like to use data "event" on ADXL345 accelerometer but it don't seems to be implemented.

In fact if i look into code (https://github.com/rwaldron/johnny-five/blob/master/lib/accelerometer.js) there is no trace of "freq".

Is it plan to develop this functionality ?

Thks

fivdi, 2017-03-11:
Hi @alex-def. The ADXL345 has "data" and "change" events implemented. I just gave them a quick try on a BeagleBone Black and they functioned correctly without any issues.

What have you tried and what were the problems?
Can you post a program which can be used to reproduce the issue?
What platform are you working on?

alex-def, 2017-03-11:
Hi Brian,

I'm using johnny-five over node-red on a raspberry pi zero.
Here my config:
```
Welcome to Node-RED
===================
11 Mar 20:09:18 - [info] Node-RED version: v0.16.2
11 Mar 20:09:18 - [info] Node.js  version: v6.9.5
11 Mar 20:09:18 - [info] Linux 4.4.38+ arm LE
```
```
  "_from": "johnny-five@latest",
  "_id": "johnny-five@0.10.6",
```

I want to get x/y data from accelerometer every second (use 10 sec on my exemple) that why i use data "event". Regarding node-red console.log, i can see x/y data but it's don't appear every 10 sec.  Lots of logs every sec then pause then logs every sec ..... 

Also if i use change "event" i ve some latency on displaying log, if i move accelerometer i ve to wait arround 10,15 sec to see log.

```
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0
[8:28:07 PM] x:  1.21 | y:  0

GAP

[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[8:28:25 PM] x:  1.21 | y:  0
[... 1983 lines cut ...]
```
var accelerometer = new five.Accelerometer({
    controller: "ADXL345",
    freq: 10000 // 10 sec
  });

  accelerometer.on("data", function() {
    console.log("accelerometer");
    console.log("  x            : ", this.x);
    console.log("  y            : ", this.y);
   
    var d = new Date();
    console.log( d.toLocaleTimeString());
    
    console.log("--------------------------------------");
  });
```

Here node-red code:

`[{"id":"aea8764a.3593e8","type":"johnny5","z":"600dd14b.5839e","name":"ADXL 345","func":"var accelerometer = new five.Accelerometer({\n    controller: \"ADXL345\",\n    freq: 10000 // 10sec\n    // Optionally set the range to one of\n    // 2, 4, 8, 16 (±g)\n    // Defaults to ±2g\n    // range: ...\n  });\n\n  accelerometer.on(\"data\", function() {\n    var d = new Date();\n    console.log(\"[\" + d.toLocaleTimeString() + \"] x: \", this.x, \"| y: \", this.y );\n  });","board":"317b390a.bffad6","noerr":0,"x":400.8333282470703,"y":193.00001525878906,"wires":[[]]},{"id":"317b390a.bffad6","type":"nodebot","z":"","name":"raspberry","username":"","password":"","boardType":"raspi-io","serialportName":"","connectionType":"local","mqttServer":"","socketServer":"","pubTopic":"","subTopic":"","tcpHost":"","tcpPort":"","sparkId":"","sparkToken":"","beanId":"","impId":"","meshbluServer":"https://meshblu.octoblu.com","uuid":"","token":"","sendUuid":""}]`

Hope you have all necessary infos.

Thks

fivdi, 2017-03-11:
The `freq` option doesn't appear to be available for the ADXL345 I2C acclerometer, perhaps it's only available for analog accelerometers like the ADXL335. The documentation for the accelerometer [data event](http://johnny-five.io/api/accelerometer/#events) which implies that all accelerometers have a `freq` option appears to be incorrect.

I'm not familiar with Node-RED so I'm guessing that you took the formatted JavaScript code posted above, modified it a little, and mapped it to the unformatted Node-RED code also posted in the above post. if this is the case, perhaps the following code or a variant thereof could be used with Node-RED
```js
var five = require('johnny-five');
var RaspiIO = require('raspi-io');

var board = new five.Board({
  io: new RaspiIO()
});

board.on('ready', function() {
  var accelerometer = new five.Accelerometer({
    controller: "ADXL345"
  });

  setInterval(function () {
    var d = new Date();

    console.log("[" + d.toLocaleTimeString() + "] x: ", accelerometer.x, "| y: ", accelerometer.y );
  }, 1000);
});
```
This program prints the x/y data from the accelerometer to the console once per second without using any events.

fivdi, 2017-03-12:
@alex-def One additional point. Above you mention a lag of 10 to 15 seconds when displaying results if the `change` event is used.

> Also if i use change "event" i ve some latency on displaying log, if i move accelerometer i ve to wait arround 10,15 sec to see log.

This is highly likely to be related to the way `console.log` is being used. `console.log` is asynchronous so it doesn't wait for messages to be logged to the console. It puts the messages in a queue and logs them to the console later. If `console.log` is called a lot, which is the case in the test program, and the messages can't be logged to the console fast enough, the queue of messages will get fuller and fuller. A visible effect of this will be lag/latency. If the program continuously logs messages to the console as fast as it can, the queue can get so full that there will be "out of memory" errors (see https://github.com/nodejs/node/issues/2970 for further info.)

alex-def, 2017-03-17:
Hi,
I've tried using **setInterval** func to simulate data "event" but this func is unfortunatly imcompatible on node-red. (So we can conclude that johnny five  event "data" will never works on node-red ;) )

I've found a way to bypass **event data** and **setInterval**  func by using this code into onReady func:

```
// Overload setInterval func to be compatible with node-red
function interval(func, funcArg, wait, times) {
    var interv = function() {
        if (typeof times === "undefined" || times-- > 0) {
            setTimeout(interv, wait);
            try {
                func.call(funcArg);
            }
            catch(e) {
                times = 0;
                throw e.toString();
            }
        }
    };
    setTimeout(interv, wait);
}

var accelerometer = new five.Accelerometer({
    controller: "ADXL345"
});

interval(function(){
    
     var d = new Date();
     console.log("[" + d.toLocaleTimeString() + "] x: ", accelerometer.x, "| y: ", accelerometer.y );
    },
    accelerometer,
    500
);
```

Here my node-red code block:

`[{"id":"aea8764a.3593e8","type":"johnny5","z":"600dd14b.5839e","name":"ADXL 345","func":"// Overload setInterval func to be compatible with node-red\nfunction interval(func, funcArg, wait, times) {\n    var interv = function() {\n        if (typeof times === \"undefined\" || times-- > 0) {\n            setTimeout(interv, wait);\n            try {\n                func.call(funcArg);\n            }\n            catch(e) {\n                times = 0;\n                throw e.toString();\n            }\n        }\n    };\n    setTimeout(interv, wait);\n}\n\nvar accelerometer = new five.Accelerometer({\n    controller: \"ADXL345\"\n});\n\ninterval(function(){\n    \n     var d = new Date();\n     console.log(\"[\" + d.toLocaleTimeString() + \"] x: \", accelerometer.x, \"| y: \", accelerometer.y );\n    },\n    accelerometer,\n    500\n);\n","board":"317b390a.bffad6","noerr":0,"x":388.8333282470703,"y":113.00001525878906,"wires":[[]]},{"id":"317b390a.bffad6","type":"nodebot","z":"","name":"raspberry","username":"","password":"","boardType":"raspi-io","serialportName":"","connectionType":"local","mqttServer":"","socketServer":"","pubTopic":"","subTopic":"","tcpHost":"","tcpPort":"","sparkId":"","sparkToken":"","beanId":"","impId":"","meshbluServer":"https://meshblu.octoblu.com","uuid":"","token":"","sendUuid":""}]`

Thks for your help :)

fivdi, 2017-03-17:
> I've tried using setInterval func to simulate data "event" but this func is unfortunatly imcompatible on node-red.

I'm not 100% sure what func is being referred to here. setInterval or the event handler.

> (So we can conclude that johnny five event "data" will never works on node-red ;) )

But the data event does work with Node-RED. The event handler for the data event resulted in a huge amount of information being logged to the console as shown in a post above. This is the correct behavior. Granted, it's not what you wanted, but that's a different issue.

On the other hand, it's good to hear that things are somehow working now.

rwaldron, 2017-03-20:
Thanks @fivdi!

> The documentation for the accelerometer data event which implies that all accelerometers have a freq option appears to be incorrect.

This is 100% correct! It looks like the description of the `data` event was just copy/pasted and that was likely my fault. 

@all 

`freq` is not available for any accelerometer, because it doesn't really make sense to allow artificially limiting the resolution of acceleration (here's an awesome accelerometer explainer: https://learn.sparkfun.com/tutorials/accelerometer-basics). If a program wants to limit the number of measurements that are collected, the `setInterval` function would be appropriate: 

```js
var accelerometer = new five.Accelerometer({
    controller: "ADXL345"
});

setInterval(() => {
  console.log(accelerometer.x, accelerometer.y);
}, 10000); // 10 seconds
```

dtex, 2018-01-02:
It sounds like @alex-def has a workaround and @rwaldron updated the documentation back on 3/20. Is it safe to close this issue?

closed 2019-01-04 by stale[bot]