rwaldron/johnny-five #1304: Joystick axes have the same value

NicolasThebaud, 2017-03-12:
Hi !

I wrote a little script to test a joystick I salvaged from an old controller.
It just reads the value from two digital pins and prints it in the console.
```
board.on('ready', function() {
	var joystick = new j5.Joystick({
		pins: ['A0', 'A2']
	})

	joystick.on('change',  function() {
		console.log('\033c')
		console.log("****** Joystick ******")
		console.log('x : ', this.x)
		console.log('y : ', this.y)
		console.log("**********************")
	})
})
```
The only problem is that the Y axis' value & X's are the same

here's how the whole thing's wired:
[image]
Red jumpers : power
White : ground
Green/Blue : data

The six jumpers on the right are wired to the Joystick as follows:
[image]

Is my joystick broken ? Did I do something wrong ?

rwaldron, 2017-03-30:
The images aren't showing for me :|

dtex, 2017-08-17:
Hi @NicolasThebaud 

It's been a while with no response so I'm going to go ahead and close this. If you are still having a problem feel free to re-open but please add images of your wiring.

closed 2017-08-17 by dtex