rwaldron/johnny-five #1305: Error: Writing to COM port (GetOverlappedResult): Unknown error code 3     at Error (native)

arunyka, 2017-03-18:
When i am going to run javascript file that uses leapjs and johnny-five i got following error

Error: Writing to COM port (GetOverlappedResult): Unknown error code 3
    at Error (native)

in the code  i am trying to rotate servos attached to arduino uno, with the help of johnny-five. how to fix this ?

dtex, 2017-03-18:
Hi @arunyka How are you powering your servos?

arunyka, 2017-03-19:
i am using power adapter that is connected to the uno board. from the uno i m giving power to the breadboard. six servos(MG 996R)  getting power from the breadboard.

rwaldron, 2017-03-30:
Can you provide a clear picture of your hardware?

@reconbot does the error above look familiar to you?

reconbot, 2017-03-31:
`GetOverlappedResult` has to do with reading, I can't say for sure that it's indicative of anything in particular.

rwaldron, 2017-03-31:
@reconbot ok, thank you! 

@arunyka picture?

arunyka, 2017-04-01:
Hi all, 

The error was fixed. I gave power to six servos directly by passing uno board.  Thanks all :)

closed 2017-04-01 by arunyka