rwaldron/johnny-five #1307: Feature Request: 433 Mhz tx/rx support

danielhep, 2017-03-21:
There are numerous cheap 433 Mhz receiver and transmitter pairs available on [eBay](http://www.ebay.com/itm/433Mhz-RF-Transmitter-and-Receiver-Module-link-kit-for-Arduino-USA-seller-/122345523978?hash=item1c7c5c8b0a:g:24kAAOSwHaBWlXJ8). A common use for them is controlling remote controlled outlets like [these](https://www.amazon.com/Etekcity-Wireless-Electrical-Household-Appliances/dp/B00DQELHBS/ref=sr_1_1?ie=UTF8&qid=1490077232&sr=8-1&keywords=etekcity+remote+controlled+outlets) ones. 

Currently the only way I can think of to control these is to use the [rcswitch-pi](https://github.com/r10r/rcswitch-pi) library and issue shell commands to it from within Johnny-Five, which isn't ideal. 

There's a complete library available [here](https://github.com/sui77/rc-switch) which works very well for the Arduino. It would be fantastic to see this implemented in Johnny-Five.

soundanalogous, 2017-03-21:
Support for those radios cannot be added to johnny-five for Arduino unless they are supported by Firmata first. Someone would need to write a Firmata wrapper for the rc-switch library and add it to [ConfigurableFirmata](https://github.com/firmata/ConfigurableFirmata) as a contributed feature. Once that is in place, it could be implemented in johnny-five.

danielhep, 2017-03-23:
@soundanalogous Okay, so should I create an issue under ConfigurableFirmata?

closed 2017-03-23 by danielhep