rwaldron/johnny-five #1309: Feature Request: Allow fading to a specified color over specified time with RGB Led.

danielhep, 2017-03-23:
Similar to the fade method for standard LEDs, a fade to color method would be perfect for RGB LEDs. This would make doing smooth color effects much easier.

dtex, 2017-03-23:
This might already be doable with the Animation class, though it's not well documented.

````
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {

  var led = new five.Led.RGB({
    pins: { red: 6, green: 5, blue: 3 }
  });

  var animation = new five.Animation(led);

  animation.enqueue({
    duration: 2000,
    cuePoints: [0, 1.0],
    keyFrames: [ [255, 0, 0], [0, 0, 255] ]
  });

});
```` 
Even if this works, I acknowledge that a wrapper method like ```Led.fade()``` would be nice.

dtex, 2018-01-11:
Hi @danielhep ,

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this feature yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for Led.RGB.fade there.

closed 2018-01-11 by dtex