rwaldron/johnny-five #1310: Use analog output as digital in relay pins

gokhandilek, 2017-03-29:
Hi,
I am trying to use the relay module of the johnny-five.
Everyhing works fine when i use the digital pins.
For example
relays.firstRelay = new five.Relay ( {
pin:13,
type:'NO'
});
I get 5V out no issue.
If I try pin 14, I will get 0V.

This is what I have tried so far.
relays.firstRelay = new five.Relay( {
pin:14,
type: 'NO'
});
and 
analogPinsToDigitalPins.pin14 = new five.Pin({
pin:14,
type: 'digital'
});
relays.firstRelay = new five.Pin({
pin:analogPinsToDigitalPins.pin14
});

Cannot get 5V output regardless.
Thank you.

rwaldron, 2017-03-30:
Thanks for the report, I will try to address this issue by tomorrow.

rwaldron, 2017-03-30:
Move to #1277

closed 2017-03-30 by rwaldron