rwaldron/johnny-five #1311: Firmata communication via audio cable instead of USB with WebJack

jywarren, 2017-03-29:
Hi, all -- although these are still not full-featured, they are working on ATMega 328 boards and are pretty cool (via @rmeister):

https://github.com/publiclab/webjack and https://github.com/publiclab/webjack-firmata 

I just wanted to put this out there as we've been able to communicate with Arduinos from a smartphone browser with no plugins or extensions, in JavaScript. I'm a big admirer of the johnny-five project and wanted to just give a heads up in case folks were interested in WebJack-style workflows.

We're open to ideas and contributions -- keep up the great work!

rwaldron, 2017-03-30:
Fascinating!

dtex, 2018-01-01:
I propose adding a "Cool AF" label

jywarren, 2018-01-02:
:-) I'd love to see more adoption - I've been trying to get an air quality
sensor demo working.

closed 2019-01-04 by stale[bot]