rwaldron/johnny-five #1317: Can not read serial impulse from bill acceptor

vgavdeev, 2017-04-05:
Hello! I am using Arduino with J5. It is really cool. Respect you.
Now I need to read impulse from bill acceptor Apex 5400 by Arduino. Please give me an example of code.

rwaldron, 2017-04-06:
Hello! Thanks for the kind words.

> Apex 5400

Honestly, I've never used one of these before, or even seen one for that matter, but based on the Arduino examples I've seen in the wild, I think you can just use an instance of `Digital` (which is an alias to `Sensor.Digital`)?

I'm going to try to make a simulator, based on the datasheet

rwaldron, 2017-04-06:
Ok, so based on the following data that I read in this manual: https://pyramidacceptors.com/pdf/Apex_Manual.pdf

1. Pulsing: 
  - Fast Pulse (50ms on, 50ms off)
  - Slow Pulse (50ms on, 300ms off)
2. Pulse Per Dollar can be configured to any number between 1 and 127 (or none, but I didn't attempt to simulate that).
3. "Acceptance" is marked by a 100ms

With that, I created a "simulator" program that picks a pseudo random bill amount and "fast pulses" that amount, at 1 PPD, followed by a 100ms "acceptance": 

```ino
int denominations[] = {1, 5, 10, 20, 50, 100};
int ppd = 1; // Any number between 1 and 127
int pin = 12;

void setup() {
  Serial.begin(9600);
  pinMode(pin, OUTPUT);
}

void loop() {
  int index = random(0, 5);
  int denomination = denominations[index];
  
  Serial.println(denomination);

  int pulses = denomination * ppd;

  // P. 4, Fast Pulse 50ms on, 50ms off
  for (int i = 0; i < pulses; i++) {
    digitalWrite(pin, HIGH);
    delay(50);
    digitalWrite(pin, LOW);
    delay(50);
  }
  // https://pyramidacceptors.com/pdf/Apex_Manual.pdf
  // 100msec when bill is recognized.
  delay(100);
}
```

I loaded this onto an Arduino and set up this: 

[image]

Then I used the following program to interpret the pulses as dollar amounts: 

```js
const { Board, Digital } = require("johnny-five");
const board = new Board();

board.on("ready", () => {
  const acceptor = new Digital(8);
  const denominations = [1, 5, 10, 20, 50, 100];
  let denomination = 0;
  let last = 0;
  let timeout = null;

  acceptor.on("data", () => {
    let { value } = acceptor;

    if (last !== value && value === 1) {
      if (timeout) {
        clearTimeout(timeout);
      }
      denomination++;
    }

    // If the last value was a countable pulse,
    // we MIGHT be at the end, so set a timeout
    if (last === 1 && value === 0) {
      timeout = setTimeout(() => {
        if (denominations.includes(denomination)) {
          console.log(`\$${denomination}.00`);
        }
        denomination = 0;
      }, 100);
    }

    last = value;
  });
});
```

Which produced output that looks like: 

```
$ node eg/sensor-digital-bill-acceptor.js
1491510509794 Connected /dev/cu.usbmodem1411
1491510511515 Repl Initialized
>> $20.00
$50.00
$1.00
$10.00
$1.00
$10.00
$1.00
$1.00
$5.00
$50.00
$10.00
$20.00
$50.00
$1.00
$1.00
...
```

rwaldron, 2017-05-09:
Hi! Just wondering if the code I provided above worked as shown?

rwaldron, 2017-05-31:
@vgavdeev ping

dtex, 2018-01-01:
Hi @vgavdeev Just wanted to follow up and see if you had a chance to try @rwaldron's suggestion

dtex, 2018-01-16:
Hi @vgavdeev 

We haven't heard back so I'm closing this, but I want you to know we aren't forgetting about this. I've added the Apex 5400 to our Johnny-Five [Requested Features](../wiki/Requested-Features) page. Hopefully we'll get a motivated contributor and they will come in and add support.

closed 2018-01-16 by dtex