rwaldron/johnny-five #1322: motor lib driver for Pololu VNH5019

zoobot, 2017-04-12:
Not sure if this is the right place to ask about this.
The Pololu VNH5019 motor driver is not represented in the johnny-five library but when I looked at the code, starting at line 1022, looks like Pololu has another motor driver in there.
I tried calling them about it but all the techs are on lunch right now.

Is this all the code needed for their motor shield? just pin numbers? Could I just as easily use this for my driver (VNH5019)?

Johnny-five is really cool. 
Thanks

https://github.com/rwaldron/johnny-five/blob/master/lib/motor.js

  POLOLU_DRV8835_SHIELD: {
    M1: {
      pins: {
        pwm: 9,
        dir: 7
      }
    },
    M2: {
      pins: {
        pwm: 10,
        dir: 8
      }
    }
  },

dtex, 2017-04-12:
Hi @Zoobot

The default pin mappings on your shield are different then the 8835. This shield will work with Johnny-Five, but you have to explicitly set the pin numbers. 

'''
var m1= new five.Motor([2, 4, 9]);
var m2= new five.Motor([7, 8, 10]);
'''
This shield also supports a current sensing pin and braking, though it's not yet clear to me how the braking works (tough to read github code on my phone).

zoobot, 2017-04-12:
Thanks @dtex! I will try that shortly. I am almost done wiring everything up.

zoobot, 2017-04-12:
@dtex That did not work...  tried 3 different motors just in case I got all lemons.
I am using a motion sensor too and that is working strangely with the shield. Now if I trigger it, it just keeps triggering forever after.
All is not hopeless though, at least the LED still works with the shield. I can still tell it to blink and it behaves very well! :)

Did I miss anything? (Other than server, etc)
board.on("ready", function() {
    console.log('Arduino connected');
    var led = new five.Led(13);
    var motion = new five.Motion(12);
    // var m1= new five.Motor([2, 4, 9]);
    // var m2= new five.Motor([7, 8, 10]);

    var m1 = new five.Motor({
      pins: {
        pwm: 2,
        dir: 4,
        cdir: 9
      }
    });
    var m2 = new five.Motor({
      pins: {
        pwm: 7,
        dir: 8,
        cdir: 10
      }
    });
m1.start(255);
}

dtex, 2017-04-12:
On that motor shield pin 12 is a diagnostic pin so it will collide with the Motion instance.

Try commenting out the motion code and see if your motor code works. I'm on a real computer now so I'll look at the docs for the shield and see if I overlooked something.

dtex, 2017-04-12:
What are you using to power the motors?

zoobot, 2017-04-12:
A variable power supply for electronic testing.
I talked with the Pololu tech and he said try hooking it up with their test code on Arduino to see if it works at all. I'll try that next. He had me directly power the pwm with 5v to drive it which did not work.
I moved commented out the motion sensor to no avail.
Im now on my phone, will report back once I try on Arduino.
I was so pleased with my soldering job but maybe it wasn't as good as I thought.

zoobot, 2017-04-13:
The Arduino test worked... motors and motor driver works... I'm relieved I didn't immediately fry my motor driver! 
https://github.com/photodude/DualVNH5019MotorShieldMod3/blob/master/examples/Demo/Demo.ino

m1A, M1B, M2A, M2B are getting no voltage with the johnny-five, even though there's voltage coming in VIN. It is getting +5 on M1INA, M1INA, M2INA, M2INB though.

Let me know if you think of something to try. Thanks for all your help @dtex!!!

dtex, 2017-04-13:
I'm glad the shield works. So when you call m1.start(255) you read 5v on both M1INA and M1INB? Only one of the two should read a voltage so we're on to something there.

For grins, try ```m1.forward(255)```. I don't think it will make a difference, but it's an easy thing to test, I just want to make sure the direction is being initialized with a default.

I don't have that shield, but I do have some that use cdir pins. I won't be able to get to them until this weekend, but I will run some tests when I get there. If it works as expected, I will need to get one of those boards.

zoobot, 2017-04-13:
Both m1.forward(255) and m1.start(255) show +5 on the M1INA, M2INA pins but not voltage on the M#A or M#B pins.. 

The reason it was showing on all of them with the test is because it was changing directions so would go high on M1INA then M1INB. Not at the same time.

I tried this too: http://johnny-five.io/examples/motor-enable/ with these enable pins.
Digital 6	M1EN/DIAG	Motor 1 enable input/fault output
Digital 12	M2EN/DIAG	Motor 2 enable input/fault output

I am going to pick up Ada's driver and see if I can get that to work. 

Are you in the bay area? If you are nearby, you are welcome to borrow this driver and test it.

dtex, 2017-04-13:
> Are you in the bay area?

I'm in Texas, but thank you for the offer. I've got one on the way from Pololu.

> Not at the same time.

Okay, that makes a lot more sense and is the expected behavior. Have you tried watching the voltage on the PWM pin?

> I tried this too

Those enable pins are a different animal than enable pins on other shields. The library sets them to output pins. I need to read the docs more to understand what they are supposed to do.

zoobot, 2017-04-13:
Cool, thanks! Let me know how it goes!! 

The Pololu tech said those enable pins are high by default so shouldn't need to mess with it.

PWM pin is getting nothing with Johnny-five. However, with the test its going to 3V or so.

dtex, 2017-04-21:
Hi @zoobot, 

My VNH5019 arrived today and I have it all soldered up. Probably tomorrow before I can actually test it though. I'll report back ASAP.

zoobot, 2017-04-22:
Great!! I am competing this robot on Sunday (if I finish on time). I'm still building and haven't weighed it yet but it feels heavy and I suspect the adafruit and arduino shields may be underpowered.

dtex, 2017-04-22:
Testing now. The problem is with the enable pin in J5. Should have an answer very soon.

dtex, 2017-04-22:
These are the correct signatures:

````
var m1= new five.Motor({
        pins: {
            pwm: 9,
            dir: 2,
            cdir: 4,
            enable: 6
        }
    });

    var m2= new five.Motor({
        pins: {
            pwm: 10,
            dir: 7,
            cdir: 8,
            enable: 12
        }
    });
````

and there was indeed a bug in Johnny-Five. I am the culprit. 

I'll get a PR into Johnny-Five here shortly, but in the meantime here's a quick change you can make to get it working:

In /node_modules/johnny-five/lib/motor.js change line 740 from

````
if (typeof this.pins.enable !== "undefined" && !this.enabled) {
````

to 

````
if (typeof this.pins.enable !== "undefined") {
````

zoobot, 2017-04-22:
@dtex You are the best!! Thank you so much!!

I'll add that and test ASAP!

dtex, 2017-04-24:
@zoobot Did it work?

zoobot, 2017-04-24:
I changed /node_modules/johnny-five/lib/motor.js line 740 to 
if (typeof this.pins.enable !== "undefined") {

It didn’t work on mine. Did you make any other changes?

Am I missing something?
I can’t work on this again until Wednesday but will test again then.

const five = require("johnny-five");

let board = new five.Board();

board.on("ready", function() {
    console.log('Arduino connected');
    var led = new five.Led(13);

    var m1 = new five.Motor({
      pins: {
        pwm: 9,
        dir: 2,
        cdir: 4,
        enable: 6
      }
    });

    var m2 = new five.Motor({
      pins: {
        pwm: 10,
        dir: 7,
        cdir: 8,
        enable: 12
      }
    });
    console.log('motor ahead')
    m1.start(255);

    // m1.forward(255);
    m2.start(255);
});

dtex, 2017-04-25:
```start()```` doesn't enable a motor* which seems goofy I know. Try ```forward()``` or ```reverse()```. I'm going to revisit enable pins and state management because it certainly seems like ```start()``` should enable the motor.

dtex, 2017-04-25:
... or ```m1.enable().start(255)```

rwaldron, 2017-04-25:
I'm trying to remember if or why `enable` was treated as a separate operation. I can't help but think there was some rationale that was similar to an `Led` instance's `stop` and `off` semantics, ie. `stop` doesn't implicitly mean `off`...

dtex, 2017-04-25:
The enable state is a property of the motor controller, whereas direction and speed are states of the motor. A motor controller might be disabled by an emergency cut off, proximity sensor, or something like that. It might even be disabled by a system that is completely independent of Johnny-Five (for safety). So, we have this thing that can be independent and at the very least should be able to have an independently (and easily) managed state.

There are some very valid reasons for us not to allow this independent state. For example: It's confusing AF.

Reviewing the code, a newly instantiated motor be enabled by default. I don't see why the motor was ever disabled. I can confirm that when I was testing that shield the motor was disabled. My proposed "hack" was just a hack, it was not a real fix.

zoobot, 2017-04-25:
My bad, I thought I made the change to motor.js but I suspect I made the change on the disable instead of enable.

m2.start(), m2.forward(255) and m2.reverse(255) all work for m2.

>> m2.enable().start(255) errors out and says 
>> 
>> TypeError: Cannot read property 'start' of undefined
>> 

> 

I am confused by m1 because it is hot in both channels, M1A and M1B are both +.

I must have shorted it somehow though it looks fine and and they are somewhat far apart?

I am going to have to unsolder and resolder I guess.

> 
Thanks for all your help!!

dtex, 2017-04-25:
> I don't see why the motor was ever disabled.

```if (typeof this.pins.enable !== "undefined" && !this.enabled) {```

Should have been

```if (typeof this.pins.enable !== "undefined" && !state.enabled) {```

I'll get this fixed, some tests written and a new PR in ASAP

mossfrond, 2017-07-02:
Has anyone gotten a VNH5019 shield working with Johnny Five? 
Its not clear to me reading the above, and I'm unable to get the shield going using examples /motor.js mod above.
(I have run the shield using the example Pololu sketch, and I have verified the Leonardo works w/ J5. I have also modified the example above to blink its LED, which it does.)
Thanks!

dtex, 2017-08-12:
@mossfrond Sorry for the delayed response.

Yes it works... Or rather it will once [this PR lands](#1372)

zoobot, 2017-10-03:
Did the PR go through for this? Thanks

dtex, 2017-10-03:
Yes it did. As long as you have version 0.11.5 or later you should have this change.

nofear87, 2017-11-21:
Whats the difference between stopping and braking? In Arduino lib they set bot inputs to low and reduce pwm. But why reducing pwm is useful when stopping the motor over inputs?

Another Question, for what you use enable pin? In Arduino Lib they use it just for reading the motor status (digitalRead)

dtex, 2017-11-21:
@nofear87 Which Arduino library is that?

nofear87, 2017-11-21:
@dtex: sorry I forgot to share it; [click me](https://github.com/pololu/dual-vnh5019-motor-shield/blob/master/DualVNH5019MotorShield.cpp)

dtex, 2017-11-21:
> Whats the difference between stopping and braking? 

The VNH5019 supports slow decay breaking (as opposed to coasting). I don't think we've added that into J5 for this particular device.

> for what you use enable pin
In Johnny-Five the enable pin can be pulled low (or set high for some boards) to disable the motor. This driver allows you to also use the pin to check the state of the switch but that is not supported in J5.

nofear87, 2017-11-21:
@dtex: Thank you! 

1. So using "brake" has the same affect as call "stop"? When I understand it correctly decay burns the remaining energy to stop the motor immediately. So when I set PWM to 255 the brake effect ist the strongest?

2. When I read it correctly one pin on the driver combine enable and diagnostic for one motor output. so i can simply switch betwenn analogRead and digitalWrite? Isn't the digital state lost when i switch to analogRead?

3. The shield allow me PWM operation up to 20 kHz. Can I configure pwm frequenzy via J5?

dtex, 2017-11-21:
> So when I set PWM to 255 the brake effect is the strongest?

As I understand it, yes but I haven't done a lot of testing on this. Calling brake in J5 on a CDIR motor set PWM to 255 and both inputs to High. It would be easy to add support for variable braking strength but it's not in there yet.

> So i can simply switch between digitalRead and digitalWrite?

I don't know the pin behavior well enough to know if you can just switch whenever you feel like it, and I also don't understand why you would want to use the enable pin to check state from J5 since the state is already known within your app. I think that pin would be valuable when you have two independent systems with the "other" system needing to check the state of a motor in a simple, direct way.

> Whats the difference between disabling and set m1ina, m1inb and pwm to zero?

Setting both m1ina and m1inb to low (or both to high) results in a short with motor braking if (and only if) PWM > 0. This detail is specific thought not exclusive to the VNH. On this device setting PWM to 0 disables both switches, leaving them in a floating state. 

> Can I configure pwm frequency via J5?

The firmata protocol (how Johnny-Five ultimately communicates with the Arduino) does not support setting the PWM frequency. There is probably too much risk in letting a user do that because it could disrupt other timer dependent functions. If you really need a specific PWM frequency a custom build of firmata is probably the best way to handle this. Just curious, what do you hope to gain by changing the PWM frequency?

dtex, 2018-01-01:
I'm going to close this. If there are any questions outstanding, or I did a poor job answering before feel free to re-open.

closed 2018-01-01 by dtex