rwaldron/johnny-five #1323: Fix Piezo.prototype.note (bad frequencies)

lyzadanger, 2017-04-13:
Per https://github.com/code-dot-org/code-dot-org/pull/14332

> Override johnny-five's Piezo.prototype.note because it's playing the wrong frequencies, at least in Maker Toolkit. I think this is a bug in their component and I'm working on getting it fixed upstream. In the meantime it's fairly easy to patch in a fix in our wrapper.

lyzadanger, 2017-04-13:
I plan to work on this next week (week of Apr. 17). Someone else can grab this if that's not soon enough!

islemaster, 2017-05-04:
Fixed in https://github.com/rwaldron/johnny-five/pull/1328, released in [v0.10.10](https://github.com/rwaldron/johnny-five/releases/tag/v0.10.10).

dtex, 2018-01-01:
Looks like this issue should have been closed a while back. Re-open if I'm missing something. Thanks everyone.

closed 2018-01-01 by dtex