rwaldron/johnny-five #1326: DMX512 Lighting Control

ccorcos, 2017-04-16:
Hey there, I want to control a lighting system that uses 3-pin DMX512 on my Raspberry Pi (or 
my Arduino).

Here's the light I'm planning on buying for reference (please let me know if you'd suggest otherwise):

https://www.amazon.com/Slim-LED-Par-Can-Uplighting/dp/B014JY8PH2/ref=sr_1_10?ie=UTF8&qid=1492370507&sr=8-10&keywords=led+stage+light

I'm trying to figure out how I can control the lights but I'm not finding any libraries that are specifically trying to conquer this.

The protocol is detailed here: https://en.wikipedia.org/wiki/DMX512

If this is out of scope for your project, I'd still appreciate any suggestions you might have.

dtex, 2017-06-21:
Sorry nobody has responded on this. I wouldn't call it out of scope and it certainly seems achievable. It really is just going to take somebody who is motivated and curious enough to step up and take it on.

dtex, 2018-01-11:
Hi @ccorcos ,

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this feature yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for the DMX512 protocol there.

closed 2018-01-11 by dtex

MCDELTAT, 2018-01-27:
i hate to be that guy, but there are definitely other methods of doing DMX512 on a Pi. So I would recommend you looking to those already working solutions. Some links follow. Assuming you have a USB to DMX512 box, then there are a few free software packages, I recommend QLC+ http://www.qlcplus.org

In the past I also had a good deal of success programming my own solution an Arduino and connecting to my laptop where I pulled off all kinds of magic with OSC in MaxMSP and the like. See the Arduino Docs for the multiple libraries they have. http://playground.arduino.cc/Learning/DMX