rwaldron/johnny-five #1332: Arduino auto reconnect when Nodejs running?

nhimspec, 2017-04-28:
When server running, i unplug arduino and plug arduino again but i can't control Arduino anymore. Can we reconnect Arduino with Node Server?
P/s: sorry for my bad english :D

criticalboot, 2017-04-28:
No, We can't. Once disconnected, we lost the state sync between arduino and the server. So you need to restart the server to take control over arduino again.

nhimspec, 2017-04-28:
Is there any other solution? Can u help me? thanks :D

scottgonzalez, 2017-04-28:
You shouldn't need to restart the server, but you will need to go through the initialization process for the arduino which will involve creating a new board instance and any other components such as buttons and resetting all of your in-memory state. The specifics of how you do this likely depends on your setup.

nhimspec, 2017-04-28:
can u give me a simple example ~_~?

rwaldron, 2017-04-28:
> When server running, i unplug arduino and plug arduino again but i can't control Arduino anymore. 

It's not possible to support this.

closed 2017-04-28 by rwaldron