rwaldron/johnny-five #1333: TLS2561 -> question about Value/Level

Stan92, 2017-05-01:
Hi,
Need to know if it's correct to get value over 80000 and level over 1.0?
I m using the Light controller with a TSL2561 (and a PI 3), and I get these kind of values.
From the documentation, my understanding was the level data should be between 0 & 1.
Thanks for your support as well as the library..

rwaldron, 2017-05-01:
Sounds like bugs! 

I will try to repro this tomorrow and see what I can learn. In the meantime, can you describe the conditions in which this sensor is exposed when these readings occur?

Stan92, 2017-05-02:
The sensor is behind a plank of wood inside a room (the TLS2561 sensor is the one on the top)

[image]

Stan92, 2017-05-02:
I had some doubts about the sensor itself then I tested using a python code
https://github.com/ControlEverythingCommunity/TSL2561/blob/master/Python/TSL2561.py

It seems the values look like good using this code.
I get with the same light conditions 346 Lux vs 4149

rwaldron, 2017-05-02:
Thanks for this additional information

rwaldron, 2017-05-02:
So, I can confirm that I had different numbers as well, when compared to Adafruit's own C libraries. I looked in the source and the only difference that I could identify was that they are using the 16X gain and 13ms integration time. When I change J5 to these settings, I get the same results.

Stan92, 2017-05-02:
Sorry for my ignorance, very new to the library.. what do you mean by "when I change J5"?

closed 2017-05-02 by rwaldron

rwaldron, 2017-05-03:
@Stan92 just go ahead and do `npm install johnny-five`, make sure you have version 0.10.12 and the changes are already in the library :)

Stan92, 2017-05-03:
I thank you very much... it seems ok :-) great job

rwaldron, 2017-05-03:
Woo! Excellent :D

arcadeJHS, 2020-04-17:
Sorry to bother you again on a closed topic, but I have got a newbie question!
I am experimenting with the Adafruit TSL2561, and I am not sure about the values I am reading.

Is the "this.value" the direct raw measure of lux sensed by the device?
I mean, does a record of "1000" mean I am measuring "1000 lux"?

So is it correct to assume, reading values from TSL2561 with Johnny-Five, a scale from "0" to "40.000" lux?

Thank you!