rwaldron/johnny-five #1334: Support for Omega2+ board.

nerdyzia, 2017-05-01:
Please provide feature support for omega2+ board.

https://onion.io/store/omega2p/

fivdi, 2017-05-01:
@gunderson implemented [Omega2-IO](https://github.com/gunderson/omega2-io) for the Onion Omega2 and Omega2+ recently. You could ask over there what the current status is.

fivdi, 2017-05-07:
It looks like Omega2-IO development has been discontinued. See https://github.com/gunderson/omega2-io/issues/1.

@nerdyzia Typically Johnny-Five itself isn't extended to support new boards like the Omega2+. Instead, an IO plugin is implemented to support a particular board. For example [BeagleBone-IO](https://github.com/julianduque/beaglebone-io) for the BeagleBone Black, [Raspi-IO](https://github.com/nebrius/raspi-io) for the Raspberry Pi or [Omega2-IO](https://onion.io/store/omega2p/) for the Omega2+.

I don't think there are any TODOs for Johnny-Five related to this issue.

rudacs, 2017-10-08:
https://github.com/rudacs/omega2-io-pro

dtex, 2018-01-01:
@nerdyzia Were you able to try any of those io plugins? Can this issue be closed?

closed 2019-01-04 by stale[bot]