rwaldron/johnny-five #1335: in J5 v0.10.10 ctrl+c doesn't quit when using etherport

soundanalogous, 2017-05-02:
console prints "Board Closing." but doesn't actually quit the application in Mac OS X.

Not sure the exact version where this broke. I upgraded from J5 v0.9.17 (which didn't have this use) to v0.10.10.

Using etherport v0.1.4.

monteslu, 2017-05-02:
just found this in ble-serial as well. It will probably be a problem in my other virtual serial streams.

This is because of https://github.com/firmata/firmata.js/blob/master/lib/firmata.js#L767

Serial streams need to accept the callback and call it on write:
https://github.com/mwittig/etherport-client/blob/master/index.js#L127

rwaldron, 2017-05-02:
I just updated Etherport: https://github.com/rwaldron/etherport/commit/5a99c5717ab24136266161c6eaf519f8fea73fa3

closed 2017-05-31 by rwaldron