rwaldron/johnny-five #1337: How I can set a timer for reading the temperature value ?

hieunguyen336, 2017-05-07:
As the tittle, I would like to delay 2-3 second to read the temperature value ?  This is my code: 
`setInterval(() => {
temperature.on("change", function() {
            console.log(this.celsius + "°C", this.fahrenheit + "°F");
}, 2000);
`

fivdi, 2017-05-07:
Does the following code do what's needed?
```javascript
setInterval(() => {
  console.log(temperature.celsius + "°C", temperature.fahrenheit + "°F");
}, 10000);
```

hieunguyen336, 2017-05-07:
I just want to know how delay the console will be ? However it print the values that no delay ? How can I add the delay time for reading value sir ?

fivdi, 2017-05-07:
I don't fully understand the questions.

Would you like to delay for 2-3 seconds after the programs starts and then be informed about every change in temperature as soon as it occurs?

Or would you like to read the temperature every 2-3 seconds?

Note that the example code shown in your first post isn't valid code. This is what it looks like formatted:
```js
setInterval(() => {
  temperature.on("change", function() {
    console.log(this.celsius + "°C", this.fahrenheit + "°F");
  }, 2000);
```
Can you post a complete and valid program that doesn't function and explain what it doesn't do correctly.

mwittig, 2017-05-08:
> Or would you like to read the temperature every 2-3 seconds?

In this case setTimeout should be used instead of setInterval ensure only a single change callback function is registered.

dtex, 2017-06-21:
@hieunguyen336 I imagine you've moved on already, but just in case here are solutions for logging once every three seconds, and logging once after three seconds. Please feel free to re-open if there is still an issue or question.

### Log every three seconds
````
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {

  var temperature = new five.Thermometer({
    pin: "A0",
    freq: 3000 // This will cause the data event to be emitted once every three seconds
  });

  temperature.on("data", function() {
    console.log("celsius: %d", this.C);
    console.log("fahrenheit: %d", this.F);
    console.log("kelvin: %d", this.K);
  });
});
````

### Log once after three seconds
````
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {

  var temperature = new five.Thermometer({
    pin: "A0"
  });

  setTimeout(() => {
    console.log("celsius: %d", temperature.C);
    console.log("fahrenheit: %d", temperature.F);
    console.log("kelvin: %d", temperature.K);
  }, 3000);

});
````

closed 2017-06-21 by dtex