rwaldron/johnny-five #1341: can't use GrovePi Expander and johnny-five together

outan, 2017-05-17:
```
var five = require("johnny-five");
var raspi = require('raspi-io');
var board = new five.Board({
      io: new raspi()
});

board.on("ready", function() {
    var virtual = new five.Board.Virtual(
       new five.Expander("GROVEPI")
    );

    var led = new five.Led({
        pin: "D4",
        board: virtual
    });

    led.on();
});
```

sudo node led.js
```
1494989986106 Available RaspberryPi-IO
1494989986139 Connected RaspberryPi-IO
1494989986155 Repl Initialized
 fs.js:718
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^
Error: Unknown system error -121: Unknown system error -121, write
    at Object.fs.writeSync (fs.js:718:20)
    at Bus.i2cWriteSync (/home/pi/workspace/johnny-five/raspi-io-led/node_modules/i2c-bus/i2c-bus.js:348:13)
    at I2C.writeSync (/home/pi/workspace/johnny-five/raspi-io-led/node_modules/raspi-i2c/dist/index.js:297:37)
    at RaspiIOCore.i2cWrite (/home/pi/workspace/johnny-five/raspi-io-led/node_modules/raspi-io-core/dist/index.js:743:19)
    at Expander.value (/home/pi/workspace/johnny-five/raspi-io-led/node_modules/johnny-five/lib/expander.js:1218:17)
    at Expander.value (/home/pi/workspace/johnny-five/raspi-io-led/node_modules/johnny-five/lib/expander.js:1187:20)
    at new Expander (/home/pi/workspace/johnny-five/raspi-io-led/node_modules/johnny-five/lib/expander.js:1876:10)
    at Board.<anonymous> (/home/pi/workspace/johnny-five/raspi-io-led/led2.js:9:8)
    at emitNone (events.js:91:20)
    at Board.emit (events.js:185:7)
```

**Specs**

- node v7.2.1
- npm 3.10.10
- Raspberry Pi 3
- Raspbian Jessie
- johnny-five 0.10.6
- raspi-io 7.2.4

outan, 2017-05-17:
I found this post but it still can't solve my problem.
https://github.com/nebrius/raspi-io/issues/65#issuecomment-243302125

rwaldron, 2017-05-17:
This issue appears to be coming from raspi-io-led, let's ping @nebrius

nebrius, 2017-05-17:
I'm at a MS offsite and don't really have time (or hardware) to dig into this unfortunately. Looking at the stack trace though, it seems to be some sort of I2C issue (raspi-led is only for the on-board LED, not LEDs connected to GPIO pins). This makes sense too, given that the GrovePi does everything over I2C IIRC.

nebrius, 2017-05-17:
cc @fivdi. Does this ring any bells for you?

fivdi, 2017-05-17:
@nebrius I'm afraid it doesn't ring a bell.

@outan Although johnny-five v0.10.6 and raspi-io v7.2.4 are not that old the current versions are v0.10.13 and v8.0.1 respectively. Can you try with the latest versions please?

If the error still occurs, which is very likely, please post the complete output of the following commands:
```
i2cdetect -y -r 1
i2cdetect -y -q 1
```
If the command `i2cdetect` can't be found it can be installed with:
```
sudo apt-get update
sudo apt-get install i2c-tools
```
Also, please post the complete content of file `/boot/config.txt`

Additional questions:
What operating system are you using, Raspbian Jessie?
If you are using Raspbain Jessie are you logged on as user pi?

fivdi, 2017-05-18:
The message for the error thrown is:
```
Error: Unknown system error -121: Unknown system error -121, write
```
Error 121 isn't as unknown as the above error message claims. It means that there was a ["Remote I/O Error"](http://elixir.free-electrons.com/linux/v4.4.68/source/include/uapi/asm-generic/errno.h#L103). It looks like either Node.js or libuv doesn't know about "Remote I/O Error" and says there was an "Unknown system error" instead.

This basically means that it's not possible to communicate with the I2C device for some reason. There are many potential reasons, for example, the device is not connected up correctly, the baud rate is too high, ...

nebrius, 2017-05-22:
So this is a thing: https://github.com/rwaldron/johnny-five/issues/1343#issuecomment-302909593.

There's a fix out that _may_ fix your issue too. Can you update to the latest version of raspi-i2c (one of raspi-io's dependencies) when you get a chance and let me know if you still see this issue?

sxd1140, 2017-09-03:
i got -121 when use oled
```
1504467448908 Available RaspberryPi-IO
1504467449627 Connected RaspberryPi-IO
1504467449650 Repl Initialized
fs.js:741
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: Unknown system error -121: Unknown system error -121, write
    at Object.fs.writeSync (fs.js:741:20)
    at Bus.i2cWriteSync (/home/pi/screen-test/node_modules/i2c-bus/i2c-bus.js:348:13)
    at I2C.writeSync (/home/pi/screen-test/node_modules/raspi-i2c/dist/index.js:307:37)
    at RaspiIOCore.i2cWrite (/home/pi/screen-test/node_modules/raspi-io-core/dist/index.js:744:19)
    at Oled._transfer (/home/pi/screen-test/node_modules/oled-js/oled.js:180:19)
    at Oled._initialise (/home/pi/screen-test/node_modules/oled-js/oled.js:137:10)
    at new Oled (/home/pi/screen-test/node_modules/oled-js/oled.js:108:8)
    at Board.<anonymous> (/home/pi/screen-test/app.js:17:14)
    at emitNone (events.js:110:20)
    at Board.emit (events.js:207:7)
```

```
var five = require('johnny-five');
var Raspi = require('raspi-io');
var board = new five.Board({
  io: new Raspi()
})
var Oled = require('oled-js');

board.on('ready', function () {
  console.log('ready.');

  var opts = {
    width: 128,
    height: 64,
    address: 0x3C
  };

  var oled = new Oled(board, five, opts);
});
```

```
i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- 3d -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

```
i2cdetect -y -q 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

```
node v8.2.1
npm 5.3.0
Raspberry Pi 3
Raspbian Jessie
johnny-five@0.11.6
raspi-io@8.0.1
raspi-i2c@5.0.4
root
```

fivdi, 2017-09-03:
The address shown for the display is either 0x3c or 0x3d depending on how i2cdetect is called. This shouldn't happen. The address should be the same in both cases. Both address are actually valid addresses for SSD1306 displays.

Can you post a link to the exact type of display being used please?

sxd1140, 2017-09-04:
[image]

fivdi, 2017-09-04:
What does the other side of the board look like? Is SA0 broken out on the connector? If so, should it be grounded to ensure that the I2C address is 0x3c (see [here](https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.h#L55-L57).)

Does it work if `address: 0x3D` is used rather that `address: 0x3C`?

sxd1140, 2017-09-07:
i changed another oled screen, 0x3c both, but same error.

sxd1140, 2017-09-07:
maybe the screen is broken, i changed another screen, 0x3c both either `-r` or `-q`. works!