rwaldron/johnny-five #1342: Two Motors can not controll by PWM

dieuvh, 2017-05-19:
I have a problem. Two motors is not independent.  
var Raspi = require("raspi-io");
var NanoTimer = require('nanotimer');
var WatchJS = require("watchjs")
var watch = WatchJS.watch;
var unwatch = WatchJS.unwatch;
var callWatchers = WatchJS.callWatchers;
var timerA = new NanoTimer();
var five = require("johnny-five"),
    board = new five.Board({
        io: new Raspi()
    });

board.on("ready", function() {
    var scope = this;

    this.pinMode(21, five.Pin.OUTPUT);

    this.pinMode(22, five.Pin.OUTPUT);

    this.pinMode(23, five.Pin.PWM);

    this.pinMode(24, five.Pin.PWM);

    this.digitalWrite(21, 0);
    this.digitalWrite(22, 1);
    this.pwmWrite(23, 200);
    this.pwmWrite(24, 0);

});

Two motors were stopped. It bases on the last line.

dtex, 2017-05-19:
A couple of questions:

What kind of motor controller are you using? Some controllers will invert the PWM signal when you change directions while others require that to be handled in software.

Is there a reason you are not using the [Johnny-Five motor class](http://johnny-five.io/api/motor/)?

For example:
````
board.on("ready", function() {
  var m1 = new five.Motor([21, 23]);
  var m2 = new five.Motor([22, 24]);
  m1.forward(200);
});
````

dtex, 2017-06-21:
Hello @dieuvh 

Just following up on this. Can you give me more information or did you get it working already? If you got it working can you post your solution here so others can benefit?

Thanks

dtex, 2017-08-12:
Hi @dieuvh 

I'm going to go ahead and close this since we haven't heard back and we don't have an issue that we can recreate. If you still have an issue, feel free to re-open the ticket with more info.

closed 2017-08-12 by dtex