rwaldron/johnny-five #1343: Can't find a connected GrovePi board

kotobuki, 2017-05-19:
I am trying to use GrovePi with johnny-five in the following environment, but even if I run the sample (expander-GROVEPI.js), the display will not proceed from `Looking for connected device`.

Board
- Raspberry Pi 3 Model B
- GrovePi+ (firmware version: 1.2.7)

Software
- Raspbian Jessie with PIXEL (April 2017, 4.4.50-v7+)
- node v6.10.3
- npm 3.10.10
- johnny-five v0.10.13
- [expander-GROVEPI.js](https://github.com/rwaldron/johnny-five/blob/master/eg/expander-GROVEPI.js)

```sh
$ node expander-GROVEPI.js
1495202610738 Board Looking for connected device
```

The GrovePi board seems to be recognized according to the execution results of i2cdetect. Actually, the samples of [node-grovepi](https://github.com/DexterInd/GrovePi/tree/master/Software/NodeJS) works fine.

```sh
$ i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- 04 -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
$ i2cdetect -y -q 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- 04 -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

If you have any other things to check, things to try, etc., please let me know. Thank you very much in advance.

fivdi, 2017-05-19:
The [expander-GROVEPI.js](https://github.com/rwaldron/johnny-five/blob/master/eg/expander-GROVEPI.js) example linked to above expects the Raspberry Pi to be connected to an Arduino which isn't the case here. The GrovePi is connected directly to the Raspberry Pi. In this case `raspi-io` also needs to be used. There's as small example of such a program [here](https://github.com/rwaldron/johnny-five/issues/1341#issue-229224563).

kotobuki, 2017-05-19:
@fivdi 
Thank you very much for your quick reply!

I tried with [the example](https://github.com/rwaldron/johnny-five/issues/1341#issue-229224563) and had the same error as follows.

```sh
$ sudo node test.js 
1495205611191 Available RaspberryPi-IO  
1495205611507 Connected RaspberryPi-IO  
1495205611521 Repl Initialized  
>> fs.js:787
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EIO: i/o error, write
    at Error (native)
    at Object.fs.writeSync (fs.js:787:20)
    at Bus.i2cWriteSync (/home/pi/johnny-five/node_modules/i2c-bus/i2c-bus.js:348:13)
    at I2C.writeSync (/home/pi/johnny-five/node_modules/raspi-i2c/dist/index.js:297:37)
    at RaspiIOCore.i2cWrite (/home/pi/johnny-five/node_modules/raspi-io-core/dist/index.js:744:19)
    at Expander.value (/home/pi/johnny-five/lib/expander.js:1218:17)
    at Expander.value (/home/pi/johnny-five/lib/expander.js:1187:20)
    at new Expander (/home/pi/johnny-five/lib/expander.js:1876:10)
    at Board.<anonymous> (/home/pi/johnny-five/eg/test.js:9:8)
    at emitNone (events.js:91:20)
```

To make sure, the version of raspi-io is 8.0.1 as show below:

```sh
$ npm list raspi-io
johnny-five@0.10.13 /home/pi/johnny-five
└── raspi-io@8.0.1  extraneous

npm ERR! extraneous: raspi-io@8.0.1 /home/pi/johnny-five/node_modules/raspi-io
```

fivdi, 2017-05-19:
> Thank you very much for your quick reply!

And thank your for your reply!

Can you change the following line in `/boot/config.txt`
```
dtparam=i2c_arm_baudrate=100000
```
to
```
dtparam=i2c_arm_baudrate=10000
```
and report your findings please?

fivdi, 2017-05-19:
I'm not 100% sure but think the Pi needs to be rebooted after changing `/boot/config.txt`.

kotobuki, 2017-05-19:
@fivdi 
Thank you very much for your suggestion. I did as follows:

1. changed `dtparam=i2c_arm_baudrate=100000` to `dtparam=i2c_arm_baudrate=10000`
2. reboot
3. run the same code again

Unfortunately, it seems that the same error occurs even after the change.

```sh
$ sudo node test.js 
1495232263765 Available RaspberryPi-IO  
:)
1495232264084 Connected RaspberryPi-IO  
1495232264097 Repl Initialized  
>> ready
fs.js:787
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EIO: i/o error, write
    at Error (native)
    at Object.fs.writeSync (fs.js:787:20)
    at Bus.i2cWriteSync (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/i2c-bus/i2c-bus.js:348:13)
    at I2C.writeSync (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/raspi-i2c/dist/index.js:297:37)
    at RaspiIOCore.i2cWrite (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/raspi-io-core/dist/index.js:744:19)
    at Expander.value (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1218:17)
    at Expander.value (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1187:20)
    at new Expander (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1876:10)
    at Board.<anonymous> (/home/pi/Desktop/scs2017/tutorial/physical-computing/j5.js:13:5)
    at emitNone (events.js:91:20)
```

As I wrote in the beginning, access with `node-grovepi` is done without any problems, so it seems that there is no problem with basic hardware setting and I2C communication. Are there any other items to check?

kotobuki, 2017-05-19:
Just to make sure, this is the complete content of file `/boot/config.txt`

```
# For more options and information see
# http://rpf.io/configtxtreadme
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
dtparam=i2c1=on
dtparam=i2c_arm=on

dtparam=i2c=on
dtparam=i2c_arm_baudrate=10000
```

fivdi, 2017-05-19:
It looks like I was barking up the wrong tree. I thought the I2C baudrate was too high but this doesn't appear to be the case.

`/boot/config.txt` looks ok.

Unfortunately I don't have a GrovePi to test with, so it may take a while to figure things out :)

As far as I can tell, the I2C write operations being performed by Johnny-Five are correct. 

Perhaps it's a timing issue. Johnny-Five performs quite a number of I2C write operations in quick succession during initialization of the GrovePi. Perhaps these operations are being performed too quickly.

The two lines of code [here](https://github.com/rwaldron/johnny-five/blob/master/lib/expander.js#L1187-L1188) are in a loop that's executed during the initialization phase.

Let's slow things down to see if it helps.

Add the following function at the top or bottom of `/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js`:
```
function sleepus(usDelay) {
  var startTime = process.hrtime();
  var deltaTime;
  var usWaited = 0;

  while (usDelay > usWaited) {
    deltaTime = process.hrtime(startTime);
    usWaited = (deltaTime[0] * 1E9 + deltaTime[1]) / 1000;
  }
}
```

Then, to slow things down a bit and provide some debug information, add the following lines at the beginning of [pinMode.value](https://github.com/rwaldron/johnny-five/blob/master/lib/expander.js#L1207-L1226) and [digitalWrite.value](https://github.com/rwaldron/johnny-five/blob/master/lib/expander.js#L1230-L1247) in `expander.js`:
```
sleepus(10000);
console.log(pin);
```

Please give this a try to see if anything changes.

kotobuki, 2017-05-20:
Thank you very much for your suggestion!

I just modified `expander.js` as follows:

```js
var Board = require("./board");
var Emitter = require("events").EventEmitter;
var util = require("util");
var nanosleep = require("./sleep").nano;
var Fn = require("./fn");
var priv = new Map();
var active = new Map();

function sleepus(usDelay) {
  var startTime = process.hrtime();
  var deltaTime;
  var usWaited = 0;

  while (usDelay > usWaited) {
    deltaTime = process.hrtime(startTime);
    usWaited = (deltaTime[0] * 1E9 + deltaTime[1]) / 1000;
  }
}

function Base() {
...
  GROVEPI: {
...
    pinMode: {
      value: function(pin, mode) {
        sleepus(10000);
        console.log(pin);
...
    digitalWrite: {
      value: function(pin, value) {
        sleepus(10000);
        console.log(pin);
...
```

The result was as follows:

```
1495238924766 Available RaspberryPi-IO  
1495238925017 Connected RaspberryPi-IO  
1495238925030 Repl Initialized  
>> D2
fs.js:787
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EIO: i/o error, write
    at Error (native)
    at Object.fs.writeSync (fs.js:787:20)
    at Bus.i2cWriteSync (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/i2c-bus/i2c-bus.js:348:13)
    at I2C.writeSync (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/raspi-i2c/dist/index.js:297:37)
    at RaspiIOCore.i2cWrite (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/raspi-io-core/dist/index.js:744:19)
    at Expander.value (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1234:17)
    at Expander.value (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1198:20)
    at new Expander (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1897:10)
    at Board.<anonymous> (/home/pi/Desktop/scs2017/tutorial/physical-computing/j5.js:9:8)
    at emitNone (events.js:91:20)
```

If there are points that I misunderstood, it will be appreciated if you can point out.

fivdi, 2017-05-20:
> If there are points that I misunderstood...

You understood everything correctly :)

The error is thrown when attempting to set the mode for pin D2.

The I2C write operation that sets the pin mode is called [here](https://github.com/fivdi/i2c-bus/blob/master/i2c-bus.js#L348) and looks like this:
```js
Bus.prototype.i2cWriteSync = function (addr, length, buffer) {
  return fs.writeSync(peripheralSync(this, addr), buffer, 0, length, 0);
};
```
Can you modify this code to the following, try again, and report your findings please:
```js
Bus.prototype.i2cWriteSync = function (addr, length, buffer) {
  console.log(peripheralSync(this, addr), buffer);
  return fs.writeSync(peripheralSync(this, addr), buffer, 0, length, 0);
};
```
If I'm not mistaken, the output should look like this:
```
4 <Buffer 05 02 01 00>
```
On your system the code to modify is at /home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/i2c-bus/i2c-bus.js:348:13

The GrovePi example that works for you does the same thing [here](https://github.com/DexterInd/GrovePi/blob/master/Software/NodeJS/libs/grovepi.js#L147).

fivdi, 2017-05-20:
Leave the `sleepus` code there for the time being.

kotobuki, 2017-05-20:
I just tried as you suggested and the result was like this:

```
1495263769551 Available RaspberryPi-IO  
1495263769846 Connected RaspberryPi-IO  
1495263769860 Repl Initialized  
>> pinMode: D2 1
18 <Buffer 05 02 01 00>
fs.js:787
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EIO: i/o error, write
    at Error (native)
    at Object.fs.writeSync (fs.js:787:20)
    at Bus.i2cWriteSync (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/i2c-bus/i2c-bus.js:353:13)
    at I2C.writeSync (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/raspi-i2c/dist/index.js:297:37)
    at RaspiIOCore.i2cWrite (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/raspi-io-core/dist/index.js:744:19)
    at Expander.value (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1242:17)
    at Expander.value (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1205:20)
    at new Expander (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1906:10)
    at Board.<anonymous> (/home/pi/Desktop/scs2017/tutorial/physical-computing/j5.js:10:8)
    at emitNone (events.js:91:20)
```

fivdi, 2017-05-20:
Sorry, I made a mistake.
```js
  console.log(peripheralSync(this, addr), buffer);
```
Should be:
```js
  console.log(adds, buffer);
```

kotobuki, 2017-05-20:
I tried again, and the result was as follows (as you expected):

```
1495265415674 Available RaspberryPi-IO  
1495265415953 Connected RaspberryPi-IO  
1495265415966 Repl Initialized  
>> pinMode: D2 1
4 <Buffer 05 02 01 00>
fs.js:787
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EIO: i/o error, write
    at Error (native)
    at Object.fs.writeSync (fs.js:787:20)
    at Bus.i2cWriteSync (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/i2c-bus/i2c-bus.js:354:13)
    at I2C.writeSync (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/raspi-i2c/dist/index.js:297:37)
    at RaspiIOCore.i2cWrite (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/raspi-io-core/dist/index.js:744:19)
    at Expander.value (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1241:17)
    at Expander.value (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1204:20)
    at new Expander (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1905:10)
    at Board.<anonymous> (/home/pi/Desktop/scs2017/tutorial/physical-computing/j5.js:9:5)
    at emitNone (events.js:91:20)
```

fivdi, 2017-05-20:
Well, that looks good to me. I'm not sure what's going on.

The test program is currently using the raspi-io plugin. There's an alternative plugin called pi-io. Can you give pi-io a try to see if anything changes please.

kotobuki, 2017-05-20:
Sure. I replaced

```js
var raspi = require('raspi-io');
```

with

```js
var raspi = require('pi-io');
```

and tried again. First time failed as follows:

```sh
1495266304617 Available unknown  
1495266304704 Connected undefined  
1495266304731 Repl Initialized  
>> pinMode: D2 1
4 <Buffer 05 02 01 00>
fs.js:787
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EIO: i/o error, write
    at Error (native)
    at Object.fs.writeSync (fs.js:787:20)
    at Bus.i2cWriteSync (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/i2c-bus/i2c-bus.js:354:13)
    at PiIO.LinuxIO.i2cWrite (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/linux-io/lib/linux-io.js:322:9)
    at Expander.value (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1241:17)
    at Expander.value (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1204:20)
    at new Expander (/home/pi/Desktop/scs2017/tutorial/physical-computing/node_modules/johnny-five/lib/expander.js:1905:10)
    at Board.<anonymous> (/home/pi/Desktop/scs2017/tutorial/physical-computing/j5.js:10:5)
    at emitNone (events.js:91:20)
    at Board.emit (events.js:185:7)
```

However, after the second time it operates normally as follows!

```sh
1495266661781 Available unknown  
1495266661877 Connected undefined  
1495266661923 Repl Initialized  
>> pinMode: D2 1
4 <Buffer 05 02 01 00>
digitalWrite: D2 0
4 <Buffer 02 02 00 00>
pinMode: D3 1
4 <Buffer 05 03 01 00>
digitalWrite: D3 0
4 <Buffer 02 03 00 00>
pinMode: D4 1
4 <Buffer 05 04 01 00>
digitalWrite: D4 0
4 <Buffer 02 04 00 00>
pinMode: D5 1
4 <Buffer 05 05 01 00>
digitalWrite: D5 0
4 <Buffer 02 05 00 00>
pinMode: D6 1
4 <Buffer 05 06 01 00>
digitalWrite: D6 0
4 <Buffer 02 06 00 00>
pinMode: D7 1
4 <Buffer 05 07 01 00>
digitalWrite: D7 0
4 <Buffer 02 07 00 00>
pinMode: D8 1
4 <Buffer 05 08 01 00>
digitalWrite: D8 0
4 <Buffer 02 08 00 00>
pinMode: A0 2
4 <Buffer 05 0e 00 00>
pinMode: A1 2
4 <Buffer 05 0f 00 00>
pinMode: A2 2
4 <Buffer 05 10 00 00>
pinMode: D4 1
4 <Buffer 05 04 01 00>
digitalWrite: D4 1
4 <Buffer 02 04 01 00>
```

I tried deleting `sleepus (10000)` in two functions, but it still works without problems. The communication speed in I2C is the state before lowering speed as follows.

```
dtparam=i2c_arm_baudrate=100000
```

kotobuki, 2017-05-20:
Just to be sure, I created a new directory in another hierarchy, installed `johnny-five` and `pi-io`, tried again. And I confirmed that the following code works as expected.

```js
var five = require('johnny-five');
var raspi = require('pi-io');
var board = new five.Board({
  io: new raspi()
});

board.on('ready', function() {
  var virtual = new five.Board.Virtual(
    new five.Expander('GROVEPI')
  );

  var led = new five.Led({
    pin: 'D4',
    board: virtual
  });

  led.on();
});
```

fivdi, 2017-05-20:
Well, at least we've made progress.

Does everything still work after a reboot of the Pi?
And can the program be terminated and restarted multiple times without any issues?

kotobuki, 2017-05-20:
We certainly made progress! However, the problem seems to have not been solved completely.

I checked it after restart. First of all, I found that the code in the previous example crashes at a probability of about a few percent.

Next, I realized that crashing would be done every time by executing code which was made slightly complicated as follows.

```js
var five = require('johnny-five');
var raspi = require('pi-io');
var board = new five.Board({
  io: new raspi()
});

function sleep(ms) {
  console.log('sleep ' + ms);
  return new Promise(resolve => setTimeout(resolve, ms));
}

board.on('ready', function() {
  var virtual = new five.Board.Virtual(
    new five.Expander('GROVEPI')
  );

  var led = new five.Led({
    pin: 'D4',
    board: virtual
  });

  var button = new five.Button({
    pin: 'D3',
    board: virtual
  });

  button.on('down', function() {
    console.log('button down');
    led.on();
  });

  button.on('up', function() {
    console.log('button up');
    led.off();
  });

  this.on('exit', function() {
    led.off();
  });
});
```

Regarding this code, adding the same change as before (adding 10,000 microsecond sleep to `pinMode` and `digitalWrite`) in `expander.js` also crashed in the same way.

```sh
1495269663467 Available unknown  
1495269663556 Connected undefined  
1495269663578 Repl Initialized  
>> pinMode: D2 1
digitalWrite: D2 0
pinMode: D3 1
digitalWrite: D3 0
pinMode: D4 1
digitalWrite: D4 0
pinMode: D5 1
digitalWrite: D5 0
pinMode: D6 1
digitalWrite: D6 0
pinMode: D7 1
digitalWrite: D7 0
pinMode: D8 1
digitalWrite: D8 0
pinMode: A0 2
fs.js:787
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EIO: i/o error, write
    at Error (native)
    at Object.fs.writeSync (fs.js:787:20)
    at Bus.i2cWriteSync (/home/pi/Desktop/test/node_modules/i2c-bus/i2c-bus.js:348:13)
    at PiIO.LinuxIO.i2cWrite (/home/pi/Desktop/test/node_modules/linux-io/lib/linux-io.js:322:9)
    at Expander.value (/home/pi/Desktop/test/node_modules/johnny-five/lib/expander.js:1232:17)
    at Expander.value (/home/pi/Desktop/test/node_modules/johnny-five/lib/expander.js:1194:20)
    at new Expander (/home/pi/Desktop/test/node_modules/johnny-five/lib/expander.js:1893:10)
    at Board.<anonymous> (/home/pi/Desktop/test/b.js:14:5)
    at emitNone (events.js:91:20)
    at Board.emit (events.js:185:7)
```

That was a share about what I knew for a while.

fivdi, 2017-05-20:
Can you lower the baudrate from 100000 to 10000 or maybe 50000 to see if there are less or no errors?

kotobuki, 2017-05-20:
I changed the communication speed and tried it.

First, at 10,000 bps, I did not add sleep to pinMode and digitalWrite, nor did it crash within the range of 50 tests for each code.

Next, at 50,000 bps, there was never a crash in the range where 50 tests were done for each code as well.

kotobuki, 2017-05-20:
Furthermore, after returning the communication speed to 100,000 bps once, I lowered the communication speed little by little and checked it. As a result, I found that at 70,000 bps, it is as almost stable as 10,000 bps. While it is slightly different, 75,000 bps makes it unstable suddenly, so there seems to be a boundary around here.

fivdi, 2017-05-20:
Looking at the [GrovePi led_blink.js example](https://github.com/DexterInd/GrovePi/blob/master/Software/NodeJS/tests/led_blink.js) the flow of control through the code when turning the LED on is from [here](https://github.com/DexterInd/GrovePi/blob/master/Software/NodeJS/tests/led_blink.js#L20) to [here](https://github.com/DexterInd/GrovePi/blob/master/Software/NodeJS/libs/sensors/genericDigitalOutputSensor.js#L10-L18) to [here](https://github.com/DexterInd/GrovePi/blob/master/Software/NodeJS/libs/grovepi.js#L161) to [here](https://github.com/DexterInd/GrovePi/blob/master/Software/NodeJS/libs/grovepi.js#L139-L154). The code more or less ignores errors and is implemented in a way that will attempt to do the best it can in the case of errors. To me, it looks like the GrovePi developers had similar issues.

The underlying problem here is a hardware bug in the Raspberry Pi. It's doesn't support I2C clock stretching correctly. AVR microcontrollers (like on the GrovePi) run at fairly low clock speeds and it's not easy to implement code that runs without stretching the I2C clock at a baudrate of 100000.

The boundary that you are seeing somewhere around 70000 baud doesn't surprise me and it's what I'd expect under these conditions.

fivdi, 2017-05-20:
Also, I2C interrupts on AVR microcontrollers have a lower priority than all other interrupts. So the more interrupts handlers there are, the worse things get and the lower the baudrate can be. Lowering the baudrate will increase the chance of the AVR not needing to stretch the I2C clock.

kotobuki, 2017-05-20:
Thank you for the detailed information.

I tried the second sample which used the button and the LED at the same time afterwards. It never crashed (as I reported earlier), however there seemed to be a malfunction related to input, such as a `pressed` event occurring continuously even at 10,000 bps. In addition, functions that require the transmission of consecutive commands like LED's `pulse()` did not work properly.

It is very disappointing, but I am beginning to think that switching to Arduino Uno etc. via USB is better... :cry:

fivdi, 2017-05-20:
I'm afraid I haven't used a GrovePi so I'm unable to comment on any Johnny-Five related issues there may be. Maybe you could open issues for the problems you found?

What features of the GrovePi would you like to use that the Raspberry Pi doesn't directly support? For example, buttons and pulsing LED will work directly with a Raspberry Pi. Or have you already purchased sensors with the appropriate connectors for the GrovePi and need to use them?

fivdi, 2017-05-20:
@nebrius There appears to be scenarios where raspi-io has I2C issues. For example, it's unclear to me why switching from rasp-io to pi-io resulted in GrovePi functioning above. I see that raspi-i2c has been reimpemented in TypeScript. Some of the corresponding JavaScript code is no longer what it used to be, for example, in [writeSync](https://github.com/nebrius/raspi-i2c/blob/5.0.3/dist/index.js#L284-L302) the var `register` is always `undefined` which means that `writeI2cBlockSync` will never be called. Before the migration to TypeScript [the same code](https://github.com/nebrius/raspi-i2c/blob/4.0.0/lib/index.js#L321-L338) called `writeI2cBlockSync` when appropriate.

kotobuki, 2017-05-20:
@fivdi 
> Maybe you could open issues for the problems you found?

Sure. I just filed #1345

nebrius, 2017-05-21:
Oh whoops, yeah I made a porting error. I just published v5.0.4 of raspi-i2c with a fix, let me know if it works!

kotobuki, 2017-05-21:
@nebrius 
Sure. I just tried with the following code:

```js
var five = require('johnny-five');
var raspi = require('raspi-io');
var board = new five.Board({
  io: new raspi()
});

board.on('ready', function() {
  var virtual = new five.Board.Virtual(
    new five.Expander('GROVEPI')
  );

  var led = new five.Led({
    pin: 'D3',
    board: virtual
  });

  led.on();
});
```

However, the result was as follows:

```sh
$ npm list | grep "raspi-i2c\|raspi-io\|johnny-five"
├─┬ johnny-five@0.10.13
├─┬ raspi-io@8.0.1
│ ├─┬ raspi-i2c@5.0.4
│ ├── raspi-io-core@2.0.1
$ sudo node a.js 
1495333830532 Available RaspberryPi-IO  
1495333830871 Connected RaspberryPi-IO  
1495333830884 Repl Initialized  
>> fs.js:787
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EIO: i/o error, write
    at Error (native)
    at Object.fs.writeSync (fs.js:787:20)
    at Bus.i2cWriteSync (/home/pi/Desktop/test/node_modules/i2c-bus/i2c-bus.js:348:13)
    at I2C.writeSync (/home/pi/Desktop/test/node_modules/raspi-i2c/dist/index.js:307:37)
    at RaspiIOCore.i2cWrite (/home/pi/Desktop/test/node_modules/raspi-io-core/dist/index.js:744:19)
    at Expander.value (/home/pi/Desktop/test/node_modules/johnny-five/lib/expander.js:1218:17)
    at Expander.value (/home/pi/Desktop/test/node_modules/johnny-five/lib/expander.js:1187:20)
    at new Expander (/home/pi/Desktop/test/node_modules/johnny-five/lib/expander.js:1876:10)
    at Board.<anonymous> (/home/pi/Desktop/test/a.js:10:5)
    at emitNone (events.js:91:20)
```

nebrius, 2017-05-22:
Hmm...

Can you do me a favor and run:

```
sudo apt-get install i2c-tools
sudo i2cdetect -y 1
```

and report back the output of the second command?

closed 2019-02-04 by stale[bot]