rwaldron/johnny-five #1345: Single button press generates many press/release events

kotobuki, 2017-05-20:
( This is related to #1343 )

Hardware
- Raspberry Pi 3 Model B
- GrovePi+ (firmware version: 1.2.7)
- [Grove - Red LED](https://www.seeedstudio.com/Grove-Red-LED-p-1142.html)
- [Grove - Button (P)](https://www.seeedstudio.com/Grove-Button%28P%29-p-1243.html)

Software
- Raspbian Jessie with PIXEL (April 2017, 4.4.50-v7+)
- node v6.10.3
- npm 3.10.10
- johnny-five v0.10.13
- pi-io v0.2.0

When the following code is executed, although the event is issued correctly the first few times, a large number of events are issued in a short time against one button press.

At first I doubted the button chattering. However, commenting out `led.on()` and `led.off()` with the same code normally works, so it seems to be caused by software reasons.

```js
var five = require('johnny-five');
var raspi = require('pi-io');
var board = new five.Board({
  io: new raspi(),
});

board.on('ready', function() {
  var startTime = new Date();

  var virtual = new five.Board.Virtual(
    new five.Expander('GROVEPI')
  );

  // Grove Red LED
  var led = new five.Led({
    pin: 'D3',
    board: virtual
  });

  // Grove Button
  var button = new five.Button({
    pin: 'D4',
    isPulldown: true,
    holdtime: 3000,
    board: virtual
  });

  button.on('press', function() {
    console.log(new Date() - startTime + ': pressed');
    led.on();
  });

  button.on('release', function() {
    console.log(new Date() - startTime + ': released');
    led.off();
  });
});
```

```
$ sudo node test.js 
1495282460485 Available unknown  
1495282460573 Connected undefined  
1495282460596 Repl Initialized  
>> 867: pressed
885: released
899: pressed
914: released
928: pressed
2437: released
3569: pressed
4594: released
5409: pressed
5430: released
5465: pressed
5487: released
5522: pressed
5544: released
5577: pressed
5599: released
5634: pressed
5654: released
5689: pressed
5711: released
5744: pressed
5766: released
5801: pressed
5823: released
5856: pressed
5879: released
5912: pressed
5934: released
5969: pressed
5989: released
6024: pressed
6045: released
6080: pressed
6101: released
6137: pressed
6160: released
6194: pressed
6216: released
6249: pressed
6271: released
```

soundanalogous, 2017-05-20:
I've noticed this on the Pi 3 in general. The digital pins are extremely noisy. I've run into the same issue even with AndroidThings running on a Pi 3. The solution was to use a really long 100ms debounce threshold on the digital pins.

soundanalogous, 2017-05-20:
The debounce threshold in five.Button [is fixed at 7ms](https://github.com/rwaldron/johnny-five/blob/master/lib/button.js#L193). This is probably too short for the noisy Pi 3 pins. The timeout should be changed to be configurable. Try changing the 7ms value to 50 or 100 and report back if that makes any difference.

fivdi, 2017-05-20:
@soundanalogous In this case the button is not connected directly to a pin on the Pi, it's connected to a pin on a GrovePi board which communicates with the Pi via I2C. That being said, the modification you suggest may still help.

kotobuki, 2017-05-21:
Let me supplement a little about this matter. First, with `led.on()` and `led.off()` commented out, it works as expected as follows. As far as this result is seen, chattering has not occurred.

```sh
$ sudo node b.js 
1495336220453 Available unknown  
1495336220525 Connected undefined  
1495336220548 Repl Initialized  
>> 1598: pressed
1925: released
```

However, when `led.on()` and `led.off()` are enabled, multiple events occurred just by pressing and releasing the button once as shown below.

```sh
$ sudo node b.js 
1495336269718 Available unknown  
1495336269792 Connected undefined  
1495336269815 Repl Initialized  
>> 1448: pressed
1465: released
1478: pressed
1495: released
1518: pressed
1707: released
```

In addition, when I pressed the button a second time, press and release events continued to occur while I was pressing.

```sh
3942: pressed
3963: released
3998: pressed
4019: released
4053: pressed
4075: released
4134: pressed
4156: released
4190: pressed
4212: released
4246: pressed
4268: released
4302: pressed
4325: released
4359: pressed
4381: released
4415: pressed
4437: released
4471: pressed
4493: released
4527: pressed
4549: released
4583: pressed
4605: released
4639: pressed
4661: released
4698: pressed
4728: released
4765: pressed
4789: released
4823: pressed
4844: released
4878: pressed
4900: released
4934: pressed
4955: released
```

It is hard to imagine that this is due to electrical chattering, so it is speculated that something wrong happens when I2C communication occurs consecutively.

fivdi, 2017-05-21:
@kotobuki as mentioned in #1343 I don't have a GrovePi for testing so what I'm saying here is based purely on analysis of code.

The Johnny-Five and pi-io code looks good to me, provided that the firmware on the GrovePi is fast enough to handle all the interactions. I can't see anything in the Johnny-Five or pi-io code that would result in the Button and LED code interfering with each other the way they are in your setup.

On the other hand, the Node.js code provided by DexterInd for the GrovePi makes use of delays at various points in the code. For example, when reading the state of a DigitalSensor, there's a 100 millisecond delay between requesting that the state of the pin be read and actually accessing the value that was read. As can be seen [here](https://github.com/DexterInd/GrovePi/blob/master/Software/NodeJS/libs/sensors/base/digitalSensor.js#L12-L20) this is what the code looks like:
```js
DigitalSensor.prototype.read = function() {
  var writeRet = this.board.writeBytes(commands.dRead.concat([this.pin, commands.unused, commands.unused]))
  if (writeRet) {
    this.board.wait(100)
    return this.board.readByte()[0]
  } else {
    return false
  }
}
```
Johnny-Five has no such delays and does things as quickly as possible. This can be seen [here](https://github.com/rwaldron/johnny-five/blob/master/lib/expander.js#L1299-L1315):
```js
        var read = function() {
          var pinIndex = nextPinIndex();
          var pin = this.pins[pinIndex];
          var isAnalog = pin.type === "analog";

          this.io.i2cWrite(this.address, [pin.command, pinIndex, 0, 0]);
          this.io.i2cReadOnce(this.address, pin.length, function(data) {
            var value;

            if (isAnalog) {
              value = (data[1] << 8) + data[2];
            } else {
              value = data[0];
            }
            handler(pinIndex, value);
          }.bind(this));
        }.bind(this);
```
A 100 millisecond delay could also be added to the Johnny-Five code to see if it helps:
```js
        var read = function() {
          var pinIndex = nextPinIndex();
          var pin = this.pins[pinIndex];
          var isAnalog = pin.type === "analog";

          this.io.i2cWrite(this.address, [pin.command, pinIndex, 0, 0]);
          setTimeout(function(){
            this.io.i2cReadOnce(this.address, pin.length, function(data) {
              var value;

              if (isAnalog) {
                value = (data[1] << 8) + data[2];
              } else {
                value = data[0];
              }
              handler(pinIndex, value);
            }.bind(this));
          }.bind(this), 100);
        }.bind(this);
```
Note that if the suggested modification does actually fix the issue, there will almost certainly be problems when attempting to read the state of two inputs. The GrovePi has a single global variable for storing the state of a pin netween the i2cWrite and i2cRead operations. See [here](https://github.com/DexterInd/GrovePi/blob/master/Firmware/Source/v1.2/grove_pi_v1_2_7/grove_pi_v1_2_7.ino#L32) and [here](https://github.com/DexterInd/GrovePi/blob/master/Firmware/Source/v1.2/grove_pi_v1_2_7/grove_pi_v1_2_7.ino#L82). All the DesterInd code is synchronous. With Johnny-Five this is not the case. With the 100 millisecond delay above, attempts to read the state of two inputs are sure to step on each other.

kotobuki, 2017-05-21:
@fivdi 
Thank you very much for your suggestion. I added 100ms delay to `expander.js` as you suggested, however the issue was still reproducible.

However, when 10 ms sleep was added to both `i2cWrite` and `i2cReadOnce` as follows, press and release are no longer consecutive.

```js
var read = function() {
  var pinIndex = nextPinIndex();
  var pin = this.pins[pinIndex];
  var isAnalog = pin.type === "analog";

  setTimeout(function() {
    this.io.i2cWrite(this.address, [pin.command, pinIndex, 0, 0]);
  }.bind(this), 10);
  setTimeout(function() {
    this.io.i2cReadOnce(this.address, pin.length, function(data) {
      var value;

      if (isAnalog) {
        value = (data[1] << 8) + data[2];
      } else {
        value = data[0];
      }
      handler(pinIndex, value);
    }.bind(this));
  }.bind(this), 10);
}.bind(this);
```

Although it seemed as if the problem was solved by this, when I changed `led.on()` to `led.pulse()` as follows, press and release occurred consecutively again.

```js
button.on('press', function() {
  console.log((Date.now() - startTime) + ': pressed');
  led.pulse();
});

button.on('release', function() {
  console.log((Date.now() - startTime) + ': released');
  led.stop().off();
});
```

Is this due to access to I2C at another timing to execute `led.pulse()`?

fivdi, 2017-05-21:
> Although it seemed as if the problem was solved by this, when I changed led.on() to led.pulse() as follows, press and release occurred consecutively again.
> Is this due to access to I2C at another timing to execute led.pulse()?

Given that tweeking the timeouts for reading the state of a button made it possible to work around errors it wouldn't surprise me if tweeking timeouts in other places would solve the problem pulsing an LED.

To be honest, I'm not sure if this tweeking of timeouts will result in a stable solution. My gut feeling tells me it wont. @rwaldron implemented the expander for the GrovePi. Maybe he can chime in and share his thoughts on the topic.

rwaldron, 2017-05-22:
@kotobuki 

```js
var read = function() {
  var pinIndex = nextPinIndex();
  var pin = this.pins[pinIndex];
  var isAnalog = pin.type === "analog";

  setTimeout(function() {
    this.io.i2cWrite(this.address, [pin.command, pinIndex, 0, 0]);
  }.bind(this), 10);
  setTimeout(function() {
    this.io.i2cReadOnce(this.address, pin.length, function(data) {
      var value;

      if (isAnalog) {
        value = (data[1] << 8) + data[2];
      } else {
        value = data[0];
      }
      handler(pinIndex, value);
    }.bind(this));
  }.bind(this), 10);
}.bind(this);
```

That likely doesn't do what you intended, which I will assume is: 

1. Wait 10 ms
2. Do a write
3. Wait 10 ms
4. Do a read

It's actually executing sort of like this: 

1. Wait 10 ms
    1. Wait for "time cost of 1 setTimeout call"
        1. Do a write
    2. Wait for "time cost of 2 setTimeout calls" (1.i + 1.ii)
        1. Do a read

This can be observed by reducing the code to the following: 

```js
var first = 0;
var diff = null;
setTimeout(function() {
  // i2c write
  first = Date.now();
}, 10);
setTimeout(function() {
  // i2c read
  diff = Date.now() - first;
}, 10);
var interval = setInterval(function() {
  if (diff !== null) {
    clearInterval(interval);
    console.log(diff);
  }
}, 10);
```

I ran this code 200 times using: 

```bash
for run in {1..200}
do
  node two-timeouts.js
done
```

I copied the output into a file which I then ran an analysis on, which produced the following results: 

| diff in milliseconds | occurrences |
| ------------------ | ------------- |
| 0 | 171 |
| 1 | 27 |
| 2 | 2 |

There is one other thing to account for: the call to `this.io.i2cWrite(this.address, [pin.command, pinIndex, 0, 0]);` will happen before the second setTimeout handler, so that also incurs a cost... BUT, wrapping that operation and the read operation in their own setTimeouts for the same amount of time cancels out. The cost of `this.io.i2cWrite(this.address, [pin.command, pinIndex, 0, 0]);` is still the only relevant cost.

I don't own a Raspberry Pi 3, should I get this thing: https://www.amazon.com/Raspberry-Model-A1-2GHz-64-bit-quad-core/dp/B01CD5VC92 ?

fivdi, 2017-05-22:
@rwaldron Yes, get that one.

The problem I see with the 10 millisecond delay between i2cWrite and i2cReadOnce is that reading the state of a pin is split into two operations that are 10 milliconds apart and a lot can happen in those 10 milloseconds. The GrovePi firmware has a single global variable called `val` for storing the state of a pin between the i2cWrite and i2cReadOnce operations. The same global `val` is used for all pins. See [here](https://github.com/DexterInd/GrovePi/blob/master/Firmware/Source/v1.2/grove_pi_v1_2_7/grove_pi_v1_2_7.ino#L32), [here](https://github.com/DexterInd/GrovePi/blob/master/Firmware/Source/v1.2/grove_pi_v1_2_7/grove_pi_v1_2_7.ino#L82) and [here](https://github.com/DexterInd/GrovePi/blob/master/Firmware/Source/v1.2/grove_pi_v1_2_7/grove_pi_v1_2_7.ino#L608). If the J5 program attempts to read multiple pins, each with their own 10 millisecond delay between i2cWrite and i2cReadOnce, the reads for the different pins will end up clobbering the value stored in the firmwares global variable `val`.

At lease this what I think will happen.

rwaldron, 2017-05-22:
Also, lol @ [this](https://github.com/DexterInd/GrovePi/blob/master/Software/NodeJS/libs/sensors/base/digitalSensor.js#L15). I wonder if they know that this will stop the entire execution environment??

rwaldron, 2017-05-22:
@fivdi seems to me that GrovePi firmware needs to be fixed/refactored first?

fivdi, 2017-05-22:
Yes, that's a rough one. To me, it also indicates that the developer had timing problems and came up with a fairly dramatic solution. I don't think the firmware was developed with async code in mind.

fivdi, 2017-05-22:
> @fivdi seems to me that GrovePi firmware needs to be fixed/refactored first?

I agree.

closed 2019-02-04 by stale[bot]