rwaldron/johnny-five #1347: Error using Configurable Firmata trough Wifi ESP8266

rikyoshiro, 2017-05-26:
This is my first try using johnny-five as my programming language for arduino and others,
I found something interesting in the net that we can open connection through an ESP8266 module to use a J5 without any serial cable. I try to compile some information from J5 example and other site to set up a thermometer using a DS18B20.
When i try to use configurablefirmata and J5 example using a serial cable it worked well but the problem comes when I try to combine it with ESP8266. Here is what I did:

Download "configurableFirmata" to my UNO,
flash the ESP8266 it with some binary code:

> (https://gist.github.com/ajfisher/5fe60fe7d8c49b3223f0)

then connect it to UNO and set my ESP8266 as an access point, 

> (https://medium.com/@imkiran/connecting-johnny-five-arduino-raspberry-pi-etc-over-wifi-to-the-pc-using-esp8266-a10348fdb300)

and then
here is what I got:

> 
> var type = board.pins.type;
>                        ^
> 
> TypeError: Cannot read property 'type' of null
>     at Function.Pins.normalize (c:\Users\Personal-PC\nodejs\node_modules\johnny-five\lib\board.pins.js:91:24)
>     at Thermometer.Board.Component (c:\Users\Personal-PC\nodejs\node_modules\johnny-five\lib\board.js:855:23)
>     at new Thermometer (c:\Users\Personal-PC\nodejs\node_modules\johnny-five\lib\thermometer.js:927:19)
>     at Board.<anonymous> (c:\Users\Personal-PC\nodejs\eg\temperature-ds18b20.js:17:23)
>     at emitNone (events.js:91:20)
>     at Board.emit (events.js:188:7)
>     at _combinedTickCallback (internal/process/next_tick.js:73:7)
>     at process._tickDomainCallback (internal/process/next_tick.js:128:9)

and here is the code that I use:
```
var VirtualSerialPort = require('udp-serial').SerialPort;
var firmata = require('firmata');
var five = require("johnny-five");
var board = new five.Board();
var sp = new VirtualSerialPort({
    host: '192.168.4.1'
});
var io = new firmata.Board(sp);
io.once('ready', function(){
  console.log('Io Connected');
  io.isReady = true;

  var board = new five.Board({io: io,repl:true});

  board.on("ready", function() {
  // This requires OneWire support using the ConfigurableFirmata
    var thermometer = new five.Thermometer({
        controller: "DS18B20",
        pin: 2
        });

    thermometer.on("change", function() {
    console.log(this.celsius + "°C");
    // console.log("0x" + this.address.toString(16));
  });
});
});
```
I am sorry I am really new to this javascript thingies :)

soundanalogous, 2017-05-26:
Which ESP8266 module are you using? The OneWire library bundled with ConfigurableFirmata supports the ESP8266 so you should be able to install ConfigurableFirmata directly on the ESP8266 and not need the Uno at all.

rikyoshiro, 2017-05-26:
Actually I use NODE MCU LUA, yea I know that we can use that single board as a main board but actually I need a lot of port and function like pwm and others. Here the link product that I use

> http://nodemcu.com/index_en.html#fr_54747661d775ef1a3600009e

soundanalogous, 2017-05-27:
Have you tried connecting the Uno directly to your computer to isolate whether or not the issue is related to OneWire or to the WiFi bridge?

rikyoshiro, 2017-05-27:
I have tried connecting uno directly to my PC using configurable firmata, and it worked. Wifi bridge? is that related? I tought my problem above is about 'device read timeout'. Never check about that.

soundanalogous, 2017-05-27:
You have several layers of complexity so it's helpful to isolate exactly which one is causing the issue:

1. OneWire in johnny-five (try using the thermometer with the Uno connected to your computer) 
2. ESP8266 Wi-Fi bridge (try blinking an LED on the Uno with the ESP8266 connected to the Uno)

Also check the amount of memory available on the Uno. If you have all of the Firmata features included in ConfigurableFirmata you will be very near the Uno's memory limit which will lead to issues, but it seems OK since the first test case I listed above sounds like it is working for you.

rikyoshiro, 2017-05-27:
Thanks for your advice let me try it first. I will post the result later.

dtex, 2018-01-16:
Hi @rikyoshiro 

Wanted to ping you before I close this. Do you have any results to share?

rikyoshiro, 2018-01-16:
Hi @dtex 
I still dont have any result tho. I am sorry. but you can close it

closed 2018-01-16 by rikyoshiro