rwaldron/johnny-five #1348: Pin Mode Error - Pi 3, GPS GP-20U7

avcohen, 2017-05-26:
I used `raspi-config` to disable Serial/USB Console and Enabled Serial interface. I've also added `enable_uart=1` config.txt.

Running `cat /dev/serial0` from the command like outputs a stream of data from the GPS unit and seems to be doing it's job. Nonetheless, I'm having an issue with Pin 10 receiving data from the GPS Unit. When I run my code below it throws the error:

> Error: Pin "10" does not support mode "other"
  	 	

    var five = require("johnny-five");
    var Raspi = require("raspi-io");
    var board = new five.Board({
      io : new Raspi({ enableSerial : true /* apparently required as per raspi-io docs */ })
    });
    
        board.on("ready", function() {
            const gps = new five.GPS({
                pins : {
                        rx : 'P1-10',
                        tx : 'P1-8',
                }
        });

        gps.on("data", () => {
                console.log("position");
                console.log(" latitude  : ", this.latitude);
                console.log(" longitude : ", this.longitude);
                console.log(" altitude  : ", this.altitude);
                console.log("———————————————————————————————————");
        });

    });

rwaldron, 2017-05-31:
cc @nebrius @dtex

nebrius, 2017-05-31:
See https://github.com/nebrius/raspi-io/issues/94#issuecomment-304953213

dtex, 2017-05-31:
Answered in the rasp-io issues. Thanks @nebrius!

closed 2017-05-31 by dtex