rwaldron/johnny-five #1349: Set Default Pin Value to High

linchpinstudios, 2017-05-28:
I'm working is a CHIP and have a relay that I want to default to the switches being off. However when I initialize the board all the relays switch on until it's ready, then I switch them off (turn the pins high).

Is there a way that I can set the value for the pins to high?

fivdi, 2017-06-12:
There was a discussion related to this topic at https://gitter.im/rwaldron/johnny-five. The discussion started on Jun 11 at 09:25 (sorry, I don't know how to link to the discussion.)

In summary, here is what I think is happening. During the initialization phase of a Johnny-Five application the mode for all pins is set to output. As the same time the value of those pins is set to low. If a relay is triggered by a low value this will turn the relay on. The initialization phase takes a while to complete. By the time the user code gets around to executing the relay has been on for a while and can be turned off. There are scenarios where this is not desirable.

I'm not actually sure if the Johnny-Five relay component was written for usage with relays that are triggered by a low value. It may only be suitable for relays that are triggered by a high value.

soundanalogous, 2017-06-12:
I think the initial pin state should be configurable. I assume the reason it is the way it is now is because of Firmata, which has always defaulted digital pins to OUTPUT, LOW, whereas Arduino defaults to INPUT. There is a reason for the using OUTPUT in Firmata (and it's a long explanation that I'll avoid here). However I don't see that as a reason to enforce this for non-Arduino board as well, since J5 has grown popular beyond just Arduino boards. In the future (Firmata 3.0), there will be a change to keeping digital pins set to INPUT mode (this is complex breaking change which is why it needs to wait until Firmata 3.0).

rwaldron, 2017-06-12:
> I assume the reason it is the way it is now is because of Firmata, which has always defaulted digital pins to OUTPUT, LOW,

Confirm :)

dtex, 2018-01-01:
What's the correct path forward here? firmata.js + firmata seem to be the prototypical environment that we build for, and we don't add general functionality to io-plugins that can't already be supported with firmata. Is this on hold until firmata 3.0?

Or maybe we could just define what the method/signature will be for firmata.js when that option is available and @linchpinstudios or @sandeepmistry could go ahead and add it to node-chip-io?

soundanalogous, 2018-01-01:
> Is this on hold until firmata 3.0

Yes for Arduino compatible boards at least.

arpadcseh, 2018-07-05:
Hello @soundanalogous 
I've just faced with this issue. Do you know the firmata 3.0 release date? :)

arpadcseh, 2018-07-08:
I would love to but my C skill is very basic :/ Anyway thanks for the great firmata library! @soundanalogous 
(I'll apply this trick https://github.com/firmata/arduino/issues/166#issuecomment-330331475)
Cheers!

closed 2019-07-10 by stale[bot]