rwaldron/johnny-five #1352: Board is closing immediately.

facetime88, 2017-06-05:
Hi Johnny,

My board has been successfully 

> console.log('Arduino is ready!');	

Followed with another log immediately:

> Board is closing.

Is there something I miss here? Thanks

facetime88, 2017-06-06:
The johnny-five init above put on a class. But when I put in directly on index.html page inside body tag, I now even get error before "Board Closing"

> Uncaught TypeError: Cannot redefine property: board
    at Function.defineProperty (<anonymous>)
    at Repl.<anonymous> (E:\client\node_modules\johnny-five\lib\repl.js:108:12)
    at Array.forEach (native)
    at Repl.inject (E:\client\node_modules\johnny-five\lib\repl.js:107:20)
    at Repl.initialize (E:\client\node_modules\johnny-five\lib\repl.js:94:8)
    at Board.finalizeAndBroadcast (E:\client\node_modules\johnny-five\lib\board.js:434:17)
    at Board.<anonymous> (E:\client\node_modules\johnny-five\lib\board.js:147:18)
    at Board.ready (E:\client\node_modules\johnny-five\node_modules\firmata\lib\firmata.js:701:7)
    at Object.onceWrapper (events.js:290:19)
    at emitNone (events.js:86:13)

rwaldron, 2017-06-06:
Have you recompiled node-serialport for Electron? https://github.com/electron/electron-rebuild#electron-rebuild

closed 2017-06-06 by rwaldron

facetime88, 2017-06-07:
Yes, it has been recompiled. Before recompiled it said DLL initilization failed, not working at all.
Now it's working but closing immediately.

facetime88, 2017-06-07:
I do not know what is the cause of the problem and after frustrated for a week, today I ended up using Cylon and make it working in less then 3 minutes.

rwaldron, 2017-06-07:
> I do not know what is the cause of the problem and after frustrated for a week

I'm sorry, I've never encountered the issue myself, so I didn't know how to help.

>  today I ended up using Cylon and make it working in less then 3 minutes.

Lol, good luck with that.

Nishkalkashyap, 2019-02-13:
If anyone else was stuck on this issue, the answer for me was to disable repl loop. SOmething like this:

    const board = new five.Board({
        repl: false,
        port: 'COM3'
    });

iamisti, 2022-11-28:
@Nishkalkashyap you are a godsent. Been having this issue for 3 days straight, tried everything and only this helped!