rwaldron/johnny-five #1353: Any example working with finger print module?

lesliechw, 2017-06-05:
Hello all,

I try to find some guide or example to make johhny-five work with finger print module. Google didn't get any result.

Some advice please. Thanks in advance.

Lesz

lesliechw, 2017-06-06:
Can this API be use with Finger Print module? http://johnny-five.io/api/sensor/

Just wanna make sure I'm heading the right way, please advice. Thank you.

dtex, 2018-01-01:
Hi @lesliechw I'm sorry nobody responded before now. The Sensor class is probably not the right path. Can you tell us specifically which hardware you are using?

dtex, 2018-01-16:
HI @lesliechw Since we haven't heard back I'm going to close this issue. I have added your request to our [Requested Features](../wiki/Requested-Features) page. If you can tell us which device you were looking at that would be helpful. Right now we are targeting just the Adafruit Fingerprint Scanner.

Feel free to re-open this issue if you have more info for us.

closed 2018-01-16 by dtex