rwaldron/johnny-five #1356: BME280 gives 0-s

AttilaMak, 2017-06-23:
Hi All,

I have a Bosch BME280 sensor. I am working with Cypress's WICED platform at the moment.
With SPI. I'm tring to get the measurement results from the sensor, but it gaves me 0 as temperature.
I I1m ask back the coefficient the answer:
CHIP ID: 60
_bme280_calib.dig_T1 val: 6F0D
_bme280_calib.dig_T2 val: 6844
_bme280_calib.dig_T3 val: 32
_bme280_calib.dig_P1 val: 91FF
_bme280_calib.dig_P2 val: FFFFD6FF
_bme280_calib.dig_P3 val: BD0
_bme280_calib.dig_P4 val: 1465
_bme280_calib.dig_P5 val: 35
_bme280_calib.dig_P6 val: FFFFFFFF
_bme280_calib.dig_P7 val: 26AC
 _bme280_calib.dig_P88 val: FFFFD8FF
_bme280_calib.dig_P9 val: 10BD
_bme280_calib.dig_H1 val: 4B
_bme280_calib.dig_H2 val: 166
_bme280_calib.dig_H3 val: 0
_bme280_calib.dig_H4 val: 148
_bme280_calib.dig_H5 val: 0
_bme280_calib.dig_H6 val: 1E

So it seems there is communication on SPI bus. Maybe calibration mistake?
could somebody help me please?

Many thanks!
[Bosch_Sensorboards.zip](https://github.com/rwaldron/johnny-five/files/1097296/Bosch_Sensorboards.zip)

fivdi, 2017-06-24:
@AttilaMak I think you may have created an issue in the wrong place.

The code in `Bosch_Sensorbards.zip` is C code for accessing a BME280 sensor using SPI. 

Johnny-Five supports BME280 sensors but the code is implemented in JavaScript, run using Node.js and uses I2C. Johnny-Five doesn't support SPI yet.

Are you using Node.js and Johnny-Five?

AttilaMak, 2017-06-26:
Hi,

Yes. It seems I made some mistake. I ‘m using another board.

Sorry for that

Many thanks in advance!

További szép napot! / Have a nice day!
_________________________________
Attila Mák
Application Engineer – ESC
Arrow Electronics
H-1138, Budapest
Vaci Str 140.
Hungary

Phone: +3612887372
E-mail:  amak@arroweurope.com<mailto:amak@arroweurope.com>
www.arrow.com<http://www.arrow.com/> / www.arroweurope.com<http://www.arroweurope.com/>
[Arrow_Global_Logo]
[http://farm1.static.flickr.com/205/492481199_733a0b5929.jpg]    [http://www.concordia.edu/sitefiles/w3/Univservices/web-banner-thinkB4Uprint.png?c=19094]

[cid:image003.png@01CFE2EF.F8A8B1E0]

closed 2017-06-26 by AttilaMak